Dan-in-CA/SIP #176: All paths in generated HTML must be relative plz

ArakniD, 2017-01-30:
Hi,

I want to put this behind a nginx proxy and thus would like all paths relative.

I've already modified the /static to static in most files and its partly there.. all links need also be relative

ArakniD, 2017-01-30:
attached working patch file
[relative.txt](https://github.com/Dan-in-CA/SIP/files/738857/relative.txt)

Dan-in-CA, 2017-01-30:
Hmmm... What version of SIP are you running?
When I try to apply the patch the following errors are raised:
root@Test_rig:/home/pi/SIP# git apply relative.txt
error: sessions/eea57ed321b43bdac91d05ad82b9aedfce149e39: No such file or directory
error: static/scripts/java/svc1.8.3/home.js: No such file or directory (This indicates a very old version)
error: patch failed: static/scripts/schedule.js:218
error: static/scripts/schedule.js: patch does not apply
error: patch failed: templates/base.html:21
error: templates/base.html: patch does not apply
error: patch failed: templates/system_update.html:16
error: templates/system_update.html: patch does not apply
root@Test_rig:/home/pi/SIP# 

From the looks of the errors, this is from a pretty old version. At least some of the directories and files have not been present in recent version.

Have you seen the information about reverse proxies on the SIP wiki? There is an nginx example.

ArakniD, 2017-01-31:
ye I just got the same result =P must be a noob error. I'll try again
tomorrow! sorry

ArakniD, 2017-02-21:
OK turns out I just had conflicts which wouldn't merge and I want looking at the output. It updated to a version from a year back.

Added an extra line in the nginx config file on the wiki.

I have all my services under the one domain and sub locations for own cloud, uchiwa, transmission  and ospi.

Works like a treat now!

closed 2017-02-21 by ArakniD