Dan-in-CA/SIP #177: Feature request - sunrise /Sunset

Tony-in-Oz, 2017-02-27:
Has anyone developed a plug in to use derived sunset and sunrise as a start or end time?

Dan-in-CA, 2017-03-01:
Not yet. It is on the wish list.

I have been looking at the astral Python module:
https://pythonhosted.org/astral/module.html

It has everything we need. It's just a matter of finding the time to make the plugin.

martinpihrt, 2022-08-01:
Hi,
maybe it will help you. I built this plugin on OSPy (unfortunately not compatible with SIP, but you can use parts and modify to SIP, or use OSPy where it's done. Dan will surely work it into SIP).
OSPy
https://github.com/martinpihrt/OSPy
Plugin
https://github.com/martinpihrt/OSPy-plugins/tree/master/plugins/sunrise_and_sunset
Regards
Martin
[image]
[image]
[image]

Dan-in-CA, 2022-08-02:
Thanks Martin. It is good to hear from you.

One thing I noticed in the advice on the second page image is about water drops on leaves causing burn in the middle of the day. This is no longer considered to be true:
https://www.gardeningknowhow.com/garden-how-to/watering/watering-leaf-burn.htm

the part about evaporation is good.

Dan

martinpihrt, 2022-08-04:
Hi Dan,
thanks for the reminder about watering during the day.
best regards
Martin

nmenoni, 2022-08-18:
Hi, this question is a bit out of the main topic. In the first image posted by Martin I see a graph related to water balance. Is that feature of graphs possible in SIP? Can we graph any values? Thanks

Dan-in-CA, 2022-08-18:
It is possible to add graphing as a new feature.
There is a feature that lets you display plugin values in the page footer. That is where Maritn has his graph. There is an example of the footer display in the proto plugin.
I am working on preparing a Python3 only version of SIP as Python 2.x has reached end of life and the latest version of Raspi OS has python3 as the default and only version. Until that is finished I don't plan to add new features.