Dan-in-CA/SIP #178: Unable to run an interval program every day

chrisbaer10, 2017-03-10:
I added an interval of 1 day in modify.html, but when I save it, but it saves as "Odd Days Only".  

I would like to run this program: 
Starting at 12:00AM until 11:59PM every 1 day, run the sprinker for 15 minutes every 1 hour.

Doesn't seem like this is possible within the current configurations... but I could be wrong.

Dan-in-CA, 2017-03-10:
Hi Chris,
It sounds like you should use a "weekly" program rather than an Interval. Set the program to run every day and you can set the start time, stop time duration and interval that you want.

[image]

chrisbaer10, 2017-03-10:
Wow, that was really easy... thanks for the help!  I now have both my aquaponics water pump and my garden drip irrigation running on the schedule I had previously scheduled in crontab.  This is a much better solution, and it adjusts for weather conditions! 

Awesome software, keep up the great work!

closed 2017-03-10 by chrisbaer10