Dan-in-CA/SIP #179: Does this code run on BeagleBone Black?

leandrojcaetano, 2017-03-29:
I'm trying to run it but it returns me a lot of errors.

Dan-in-CA, 2017-03-29:
SIP has been tested on BBB running debian OS.

What sort of errors are you getting?

leandrojcaetano, 2017-03-29:
Thanks for the answer. I use this same combination (BBB + Debian). 

I received the following error:

> debian@beaglebone:~/SIP$ sudo python sip.py

> ALARM from Example_Sender!: Just and example!
> plugins loaded:
>   mobile_app
>   plugin_manager
>   proto
>   signaling_examples
>   system_update
> Starting timing loop

> zones changed
> [0, 0, 0, 0, 0, 0, 0, 0]
> http://0.0.0.0:80/
> Traceback (most recent call last):
>   File "sip.py", line 240, in <module>
>     app.run()
>   File "sip.py", line 186, in run
>     return web.httpserver.runsimple(func, ('0.0.0.0', port))
>   File "/home/debian/SIP/web/httpserver.py", line 157, in runsimple
>     server.start()
>   File "/home/debian/SIP/web/wsgiserver/__init__.py", line 1753, in start
>     raise socket.error(msg)
> socket.error: No socket could be created

leandrojcaetano, 2017-03-29:
Hi Dan,
I found the solution in the official forum:
[http://nosack.com/sipforum/showthread.php?tid=7](http://nosack.com/sipforum/showthread.php?tid=7
)

Thank you very much!

Dan-in-CA, 2017-03-29:
Glad you fund the solution.
The only actual error was that the default web socket port was already being used.

closed 2017-04-04 by leandrojcaetano