Dan-in-CA/SIP #185: Adding image resets all other station names

ntewinkel, 2017-05-22:
Software version: 3.1.53 (2016-08-19)
Using the Android App: OpenSprinkler 1.6.0

I had all 13 zones set up with custom names, but none had images.
Decided to set an image for zone 13. Took a picture, clicked "submit", and the names for zones 1-12 became the defaults S01, S02, etc. Zone 13, now with image, kept the proper name.

ntewinkel, 2017-05-22:
I'm wondering if this is specific to the Android app, as I don't see any reference to images in the web front end of this Python app.

Dan-in-CA, 2017-05-22:
Yes that would be specific to the Android app.
The repository for that is:
https://github.com/OpenSprinkler/OpenSprinkler-App

The newer versions of the SIP web interface include Bootstrap and will work on mobile devices without the Android app. Just point the browser on your mobile device to the SIP URL. However The SIP interface does not include images at this point.

For secure remote access you can use a VPN which can be easily set up with piVPN:
http://www.pivpn.io/

ntewinkel, 2017-05-22:
Thanks for the quick reply, Dan!
I've opened an issue there (https://github.com/OpenSprinkler/OpenSprinkler-App/issues/36)

I'll try the browser based way, I didn't realize that was possible.

closed 2017-05-22 by ntewinkel