Dan-in-CA/SIP #189: SIP for Orange Pi One

Riaan111209, 2017-06-03:
Hi Dan
I see there is somebody on this page that had questions about this in the past. I also have a Orange Pi One.
I want to be able to switch 21 channels/relays with it but SIP will not start. I get a message in the terminal that it is not a raspberry pi.
Can you please help me set this up. I have jumped through a lot of different irrigation stuff and so far this software looks like it is what I need for my system.
Hope to hear from you soon.
Kind Regards

Dan-in-CA, 2017-06-03:
Welcome Riaan

I don't have much time right now but I am willing to help you get SIP running on your Orange Pi. You will need to make some changes to the SIP Python code. I would suggest that you fork SIP to your GitHub repository. You can make changes there and when things are working, your changes can be merged into the main SIP repo.

It looks like the Orange Pi has the same GPIO pin layout as a RPI B+. That will make things fairly easy.

SIP detects what kind of board it is running on by trying to import a GPIO library either for the Pi or Beagle bone black. I assume you have an importable GPIO library on the Orange Pi. You can try adding it to the import instruction on line 6 of gpio_pins.py in the SIP directory for sxample:
`
import RPi.GPIO or your_library_here as GPIO
`
That should at least allow sip to run.

If you plan to  use 21 channels you should probably put together a shift register interface as described on the SIP wiki:
https://github.com/Dan-in-CA/SIP/wiki/Relay-interface

That' all I have time for right now. I will be back online later today.

Dan

Riaan111209, 2017-06-03:
Hi Dan
Thank you very much for the reply and that you will help me to get SIP
running on an orange pi.
I don't know what time it is where you are. At the moment it is 23:05 here.
I hope you can help me to set this up because $ am new to this coding and
still have a lot to learn.

Kind Regards
Riaan

Dan-in-CA, 2017-06-03:
OK.
I will work on modifying the Python code for SIP to try and get it to work with the Orange Pi.
You will need to install a Python GPIO library on your Orange Pi. Follow the instructions for "installing Python and Libraries" at:
https://diyprojects.io/orange-pi-onelite-tutorial-use-gpio-python-pinouts/#.WTNH09y1sdU

Sip should be installed into a directory named "pi". Does your Orange Pi have a pi directory? I don't have an Orange Pi so I will probably ask a few questions like that.

According to the time stamp on your previous post it looks like your local time is about 12 hours and 45 minutes ahead of the time here in California. You must be on the other side of the world.

Dan

Riaan111209, 2017-06-04:
Good morning Dan
I tried to attach 2 pictures but then my message got rejected on your side
because of the size.

Inside the code of gpio_pins.py
Line 7 says platform 'pi'
Should I change this to 'root' as that is where the SIP folder is?
Kind Regards

On 04 Jun 2017 08:18, "Riaan Engelbrecht" <riaan.eng@gmail.com> wrote:

Good morning Dan
I attached 2 pictures in this mail so that you can see what it looks like
on my side.
Thank you very much for being willing to help me with this.

Kind Regards
Riaan

https://diyprojects.io/orange-pi-onelite-tutorial-use-gpio-p
ython-pinouts/#.WTNH09y1sdU

Dan-in-CA, 2017-06-04:
Riaan,
It would be best to make a directory /home/pi and place SIP there. There is a script that is used to start SIP automatically after a reboot that looks for that location.

I will make some code changes in a branch of the SIP repository so you will need to clone a fresh version when that is ready.

Dan

Riaan111209, 2017-06-04:
Hi Dan
Thank you for the reply.
If I open the file manager then I am inside root.
Do you want me to make a new folder inside root with the name 'pi' and the
put the SIP folder inside it or does it look for "home" as well.

Kind Regards
Riaan

Dan-in-CA, 2017-06-04:
The script looks for /home/pi so you should make a home directory with a pi sub directory and place SIP there.

Riaan111209, 2017-06-04:
Thank you. I thought that might be the case.
I will do that and then see if it will run.

Kind Regards
Riaan

Dan-in-CA, 2017-06-04:
I don't think it will run yet. 
I am working on the code modification but it may take a while.

Did you install the orangepi_PC_gpio_pyH3 Python library? That will need to be ready on your system before SIP can be used.

Riaan111209, 2017-06-04:
I tried to install the python library fot gpio as you suggested but it does
not work. I am downloading a different operating system to see if I can get
it done.

Dan-in-CA, 2017-06-04:
I pushed a new branch named "orange" to the SIP repository. When you are ready to test SIP use
`git sheckout orange` form the command line in the SIP directory. That should switch to the new branch.

The orange branch has a modified version of gpio_pins.py that should work with the orange pi as long as the Python GPIO library is installed. I don't have a way to actually test this but it runs on my Raspberry Pi test rig so I don't see any major problems.

Riaan111209, 2017-06-04:
Thank you very much Dan.
I appreciate your help and support.
I will give you another reply when I have the new OS loaded.

Kind Regards
Riaan.

Riaan111209, 2017-06-05:
Good morning Dan
The link for the duxing gpio library is not working.
It gives me a 403 url error.

When I run the python dev install it said that the newest version is
already installed.

I will try and put SIP on and let you know from there.

Kind Regards
Riaan

On 04 Jun 2017 19:26, "Riaan Engelbrecht" <riaan.eng@gmail.com> wrote:

Thank you very much Dan.
I appreciate your help and support.
I will give you another reply when I have the new OS loaded.

Kind Regards
Riaan.

Riaan111209, 2017-06-05:
At the moment
The SIP folder is in the following directory way
/root/home/pi/SIP
I had to create the home and the pi folder because it installs the SIP
folder directly into root

The sip.py will not run.

Dan-in-CA, 2017-06-05:
How are you trying to start SIP?
Using the command line from in the SIP directory it should be:
`sudo python sip.py`
Do you see any error message?

Riaan111209, 2017-06-05:
Hi Dan
I use the LX terminal
Following:
cd /home/pi/SIP
Then sudo python sip.py

Then there is a few errors and the last line tell me that this will only
run on a raspberry pi.

My OS is rasbian for orange pi one

Dan-in-CA, 2017-06-05:
That is strange.
SIP does not print a message that it will only run on a Raspberry Pi. In fact SIP does also run on Beagle Bone and I have even run it on Windows. The only time I have seen that message is if I try to load the RPi.GPIO library on a non-Raspberry Pi system. 

Try running the command "git branch" in the SIP directory. You should get a reply of "orange", If it returns "master" use the command "sudo git checkout orange" That should change to the code modified for Orange Pi. If you are on the orange branch and still get the message that it will only run on a Raspberry Pi I will make some further changes to the code to eliminate any reference to RPi.GPIO.

Also, it would be helpful to know what other error messages you get when you try to tun SIP. That would help figure out what is happening.

Riaan111209, 2017-06-05:
Hi Dan
Thank you.
I will try that in about 30minutes.
If it give me all the errors then I will type everything in a message. It
would be easier to send a picture but the servers reject my email if a
picture is attached.
Will let you know.
Thanks

Kind Regards
Riaan

Riaan111209, 2017-06-05:
Okey Dan
It replied * master
I did as you said and then I got the message:

*Branch orange set up to track remote branch orangefrom origin.*
*Switched to a new branch 'orange'*

I then run the following:
*root@orangepi:~/home/pi/SIP# sudo python sip.py*

This is the terminal display after that command:
*BAPI: revision(-1)*
*Trackback (most recent call last):*
*   File "sip.py", line 17, in <module>*
*        import gv*
*   File "/root/home/pi/SIP/gv.py", line 106, in <module>*
*        from helpers import load_programs, station_names*
*   File "/root/home/pi/SIP/helpers.py", line 27, in <module>*
*        from gpio_pins import GPIO, pin_rain_sense, pin_relay*
*   File "/root/home/pi/SIP/gpio_pins.py", line 6, in <module>*
*        Import RPi.GPIO as GPIO*
*RuntimeError: This module can only be run on a Raspberry Pi!*

I hope this give you a better idea as to what is going on here at my side.

Kind Regards

Dan-in-CA, 2017-06-05:
Yes, That tells me a lot.

First, you need to actually cd into the /home/pi/SIP directory before you try to start SIP. That is what I meant by "from the SIP directory". SIP uses relative paths to load it's Python modules and if you try to start sip.py from a different directory they will not load. That is what is causing the first few errors.

You mentioned that your OS is rasbian for orange pi one. That gives me a clue abut the error that comes right after "Import RPi.GPIO as GPIO". It looks like whoever modified Rsapbian (the official OS for Raspberry Pi) for use on the Orange Pi forgot to remove the Python RPi.GPIO library. It should not exist on a non-Raspberry Pi system. SIP will try to load that library and if it is not found SIP will try to to load a different library or it will print a message that no GPIO library was fund and continue to run.
In this case SIP finds the RPi.GPIO library and tries to run it causing an error that SIP is not prepared to handle.

Try looking in /usr/lib/python2.7/dist-packages and see if there is a folder named RPi. If it is there delete it. That should get rid of the error and allow SIP to run.

Riaan111209, 2017-06-05:
Thank you for the reply.
This is the following I did.

*root@orangepi:~# cd /usr/lib/python2.7/dist-packages*
Then I put 'dir' to get the following:

*dbus                                       pifacecommon-r.0.0.egg-info*
*_dbus_bindings.so              pifacedigitalio*
*_dbus_glib_bindings.so.    pifacedigitalio-3.0.4-info*
*debconf.py                           pygame*
*debconf.pyc                         pygame-1.9.1release.egg-info*
*gi                                            pyserial-2.5.egg-info*
*picamera                               python-support.pth*
*picamera-1.6.egg-info       README*
*pifacecommon                    serial*

*root@orangepi:/usr/lib/python2.7/dist-packages#*

Dan-in-CA, 2017-06-05:
OK. Looks like RPi is not in that directory but it must be somewhere.

Try:
`find / -name RPi`

Riaan111209, 2017-06-06:
Good morning Dan
If I type *sudo find / RPI *then it run the search and end with the
following line:
*find: 'RPI': No such file or directory*

Riaan111209, 2017-06-06:
Hi Dan
For which orange pi OS did you test your system?
I might be a good idea for me to load the same operating system and start
again.

Kind Regards

Dan-in-CA, 2017-06-06:
When I try "find /  RPi" it returns "RPI': No such file or directory" but when I try "find / -name RPi" with the -name switch it finds it.

I am running the latest Raspi for Raspberry Pi so it might not work on an Orange Pi.

I will try to eliminate the import of RPi.GPIO in the orange branch. That should eliminate the error. I will let you know when the update is ready.

Riaan111209, 2017-06-06:
Hi Dan
Thank you.
It will be very nice if I can run the SIP on the orange pi one.
Very kind of you to go through all the effort.

Kind Regards
Riaan

Dan-in-CA, 2017-06-06:
I pushed an update to GitHub with the references to RPi.GPIO removed.
You can get the update by cd to /home/pi/SIP and use the command
`sudo git pull`
That should eliminate the error and allow SIP to run.

Once SIP is running use a web browser on your network to go to the URL of your Oange Pi. When you see the SIP login screen use the password "opendoor".

Riaan111209, 2017-06-06:
Hi Dan
I am reloading a new memorycard for the orange pi. Which operating system
do you think should work best with your system on the orange pi and all the
updates and changes that you made?

Dan-in-CA, 2017-06-06:
Because I don't have an Orange Pi, I can't say which OS euld be best.

Riaan111209, 2017-06-06:
Okey thank you. I will put debian back and start again.

Riaan111209, 2017-06-08:
Good morning Dan
Will I be able to use a *arad irt 100 *flow meter with SIP to monitor the
amount of water flow?

Riaan111209, 2017-06-08:
Hi Dan.
I think I need to try this with a raspberry pi first.
Reason for this is that I need to install the system on the farm at the end
of the month so I am running out of time. When I am back from the farm, I
will then try again with the orange pi one at my house in town.
I am building a relay board with the shift registers today to be able to
switch all the relays for the valves.
How does the SIP program with the raspberry pi 3 work with the shift
registers?

Kind Regards
Riaan

Dan-in-CA, 2017-06-08:
regarding the flow meter see issue 143
https://github.com/Dan-in-CA/SIP/issues/143
another contributor is developing a flow meter plugin for SIP.

If you follow the plan on the SIP wiki for the shift register set up it will work with SIP's native code and no plugin is required.

Riaan111209, 2017-06-08:
Hi Dan
Thank you for the reply. I hope I can get this working well. My goal is
still to make this work on the orange pi. Time is just catching me and I am
not an expert like you so for me this take a lot of time.

Riaan111209, 2017-06-09:
Hi Dan
I am starting to feel stupid because I can not even get SIP to work on a
raspberry pi 3.
I installed the SIP and I can open it by running the command
Sudo python sip.py

I built the shift register board as per the fritzing drawing on your github
but the relay only switch on channel 1.
In the Terminal I can see the change from 0 to 1 on all the chanels if I
switch them on.
But that is not what is happening with the relays.

Kind Regards

Dan-in-CA, 2017-06-09:
Not sure why that would be. Check the wiring very carefully.
What kind of relays are you using?

Riaan111209, 2017-06-09:
I have a 16ch relay module low level trigger.

Dan-in-CA, 2017-06-09:
The low level trigger is a problem with this set up. I have been working on a solution.
There is a setting under "Station Handling" on the SIP Options page named "Active-Low Relay" That check box should be checked.

The relays will all turn on when the system is starting up but should work properly once it is booted up.

Dan-in-CA, 2017-06-10:
I found a fix for the problem of the low level trigger relays switching on during power up. You need to add a 10K Ohm resistor between the wire from pin 11 and the 5v line from the Pi.

If you check the "acrive-low relay box on the SIP options page  as described above, and and add the resistor to the shift register circuit your relay board should work without a problem.

Riaan111209, 2017-06-10:
Hi Dan
Thank you for the info.
In your github about the auto restart where you explain how to copy the
sip.service, i get an error and it will not copy. I also saw that for some
reason my OS have 2 main directories. The one is "pi" and the other one is
"/".
The SIP is installed on "pi" and the systemd is on "/".

How can this work?

Once again thank you for all the help and support.

Kind Regards
Riaan

Dan-in-CA, 2017-06-10:
The "pi" directory is the home directory for the user "pi'. When you log in as pi you will be placed in that dir.
the "/" directory is the root directory. If you make changes to the / directory you must be "root" or use "sudo" when running a command.

What version of the OS are you running?
You can get the version with:
`lsb_release -a`
or
`cat /etc/*-release`

Dan

Riaan111209, 2017-06-11:
I typed
*lsb_release -a*
Then I got this
*No lsb modules are available.*
*Distributor ID: Raspbian*
*Discription: Raspbian GNU/Linux 8.0 (jessie)*
*Release: 8.0*
*Codename: jessie*

When i type
*cat /etc/*-release*
Then i get this
*PRETTY_NAME="Raspbian GNU/Linux 8.0*
*NAME="Raspbian GNU/Linux"*
*VERSION_ID="8"*
*VERSION="8 (jessie)"*
*ID=raspbian*
*ID_LIKE=debian*

Riaan111209, 2017-06-11:
Hi Dan
I manage to select the low level trigger as you said.
I got the relays to switch but they are not reacting to the command. If i
switch station one the it will switch 2 relays on. If I select station 2
then it will switch another 2 on. If I switch station 1 off then it will
switch 3 relays off.
This is a bit strange but at leasts I am moving a bit forward with this
project. I also loaded a new OS this afternoon "pixel with raspbian" this
seem to be more stable.
I am using 74HC595N shift registers.

Dan-in-CA, 2017-06-11:
Sounds like you are making progress. I have not seen the situation where more than one station is switched on unless you have a master station selected on the Options page or the system is set for concurrent mode.

Did you also add the 10k Ohm resistor to the shift register circuit? The diagram for the shift registers has been updated today to include the resistor.

Riaan111209, 2017-06-12:
Hi Dan.
I still need to put the resisters on my circuit.
Will do that this morning and try again.
Then I still need to do the rc.local to make it start automatic when the
power is cycled.

Riaan111209, 2017-06-13:
Hi Dan
If I put the 10kohm resister as per your drawing the nothing will switch on
or off.
I have another question:
Will it be possible to connect a arduino mega to the pi3 to let the mega do
the switching of the pins/relays?

Dan-in-CA, 2017-06-13:
Did you also check the "Active-low relay" box under "station handling' on the SIP Options page. 
Other users have it working with low level trigger relay boards.

It may be possible to use an Arduino to control the relays but you will need to write some arduino code.

Riaan111209, 2017-06-14:
Hi Dan.
Yes the box is checked for low voltage relay board.
I will try again this afternoon again.
I will build the shift register on a breadboard and try again.

Dan-in-CA, 2017-06-14:
It is always good to start with a breadboard. Once you have everything working you can move the circuit to a more permanent support or you can leave it on the breadboard.

Riaan111209, 2017-06-15:
Will not be able to leave it on the breadboard. Not reliable enough as the
farm is 2000km from me.
Will let you know what happen.

Dan-in-CA, 2017-06-17:
Riaan,
Another user reported a similar problem to what you described only after connecting the power for the irrigation valves (24V AC). It looks like it might be necessary to use a separate 5V power supply for the relays rather than using the 5V power from the Raspberry Pi.

I did not see this problem when using high level trigger relay boards. SIP was originally designed to work with shift registers which are the equivalent of  high level trigger.

Riaan111209, 2017-06-18:
Hi Dan
Thank you for the info.
I have high level relays as well.
If I use this without the shift registers, which gpio pins are SIP using
for the relay triggers.
I also have a seperate 5v power supply for the relays.

Kind Regards

Dan-in-CA, 2017-06-18:
If you need more than 12 stations you should probably use the shift registers with high level trigger relay boards.

If you want to try without the shift registers you will need to use the "relay_board" plugin. The documentation describing the pins to use is:
https://github.com/KanyonKris/relay_board/wiki

You can install the plugin from the SIP plugins page.

Riaan111209, 2017-06-18:
Thank you
I will install the plugin and try to get the 12 stations working then go
for 24 stations.

aocana, 2017-08-01:
Hi there!, @Dan-in-CA 

im trying to configure my orangepi zero to run SIP. 
I have downloaded last version or SIP of branch ORANGE, please, check the print out:
```

takezo@orangepizero:~/OrangeSIP/SIP$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 8.8 (jessie)
Release:	8.8
Codename:	jessie

takezo@orangepizero:~/OrangeSIP/SIP$ git branch
  master
* orange

```
I also modified the path of home directorey where is located SIP, modified into file sip.sh

The case is, when i try to run the sip, Im getting the following errors:

```
takezo@orangepizero:~/OrangeSIP/SIP$ sudo python sip.py 
[sudo] password for takezo: 
ALARM from Example_Sender!: Just and example!
plugins loaded:
  mobile_app
  plugin_manager
  signaling_examples
  system_update
Traceback (most recent call last):
Starting timing loop
  File "sip.py", line 247, in <module>

    set_output()    
  File "/home/takezo/OrangeSIP/SIP/gpio_pins.py", line 228, in set_output
    disableShiftRegisterOutput()
  File "/home/takezo/OrangeSIP/SIP/gpio_pins.py", line 169, in disableShiftRegisterOutput
    setup_pins()
  File "/home/takezo/OrangeSIP/SIP/gpio_pins.py", line 149, in setup_pins
    gpio.setcfg(pin_sr_noe, gpio.OUT)
AttributeError: 'module' object has no attribute 'OUT'

```

thank you so much to support me whith this installation,

BR!

Dan-in-CA, 2017-08-06:
The GPIO error you are getting indicates that the Python GPIO library is not being found. What GPIO library are you using?

aocana, 2017-08-07:
@Dan-in-CA 

Thank you so much for your answer.
Due that i cannot use this software, i have configured a python script with this setup, to control the system via cron:

`#import the library
from pyA20.gpio import gpio
from pyA20.gpio import port
from time import sleep

#initialize the gpio module
gpio.init()

#setup the port (same as raspberry pi's gpio.setup() function)
gpio.setcfg(port.PA6, gpio.OUTPUT)

#now we do something (light up the LED)
gpio.output(port.PA6, gpio.LOW)

#turn off the LED after 2 seconds
sleep(180)
gpio.output(port.PA6, gpio.HIGH)`

on the other hand, if I have understood correctly your question , I'm using  pyA20 library, but, I have not idea how to configure SIP  to use this library..

do i need to configure this library in any file into the SIP?

aocana, 2017-08-12:
Hi @Dan-in-CA ,

please could you take a look this last post..?

thanks!

Dan-in-CA, 2017-08-12:
Sorry for the delay. I have been very busy.

I am traveling today and should get back to you tomorrow.

Dan-in-CA, 2017-08-13:
After looking at the code you posted for the cron set up. It looks like the SIP code in the gpio_pins.py file on lines 149 through 152 should b OUTPUT instead of OUT.
I pushed an update to the repository.
Let me know if this works.

If you will be using a relay board to control the sprinkler valves that will bring another set of problems. It may require making a modified version of the relay_board plugin.

[15 more replies]

closed 2019-10-26 by Dan-in-CA