Dan-in-CA/SIP #190: Feature Request

dgbauman, 2017-06-11:
Hi Dan,

Not sure if this feature exists, as I just started looking at using OSPi.

I have two different sprinkler regions for my yard (Front and Back), with each region having a unique master valve.  It would be awesome if the master valve selection could be selected at the schedule program section of the webpage.

For example, for region 1, I would use solenoid #1 as the main solenoid for sprinkler solenoids 2, 3, 4, 5, 6, 7, 8.  Then for region 2, I would use solenoid #9 for sprinkler solenoids 10, 11, 12, 13, 14, 15, 16.

If you have any suggestions on which Python files I need to review for code changes, that would be great.

Overall, the OSPi project and scheduling project is awesome! Keep up the great work.

Thanks,

Don B.

Dan-in-CA, 2017-06-11:
I'll need to think about how this could possibly be implemented. 
There are references to the master throughout the code so it won't be straight forward.

One alternate possibility would be to run SIP in concurrent mode witch allows more that one station to be on at the same time. That can be enabled by un-checking the "sequential" box under Station Handling on the options page.

Dan-in-CA, 2019-12-30:
Cleaning up old issues.

closed 2019-12-30 by Dan-in-CA