Dan-in-CA/SIP #192: helpers.py function restart does not work when not using openrc

jharvell, 2017-07-02:
Install SIP on a system using systemd instead of openrc.  (Debian 7 on a Beaglbone Black in my case).  Observe annoying _modal_ pop-up saying

> SIP could not restart.
> Please restart the program from the command line.

Changing the language under options is a good way to trigger this.

This happens because it assumes the system is restarted via the openrc service command.  In my case I am using systemd.  So I installed the sip.service file in /etc/systemd/system instead of making an openrc service for it.

My workaround is below.  I tried to do this in a more system independent way by doing a sys.exit(os.EX_OK).  But that didn't work.

```
joey@pip:~/SIP$ git --no-pager log -n1
commit f116a9cacc18cd97aaee1bee9aa5699e3e370d1a
Author: Dan <dkimberling@att.net>
Date:   Fri Jun 30 08:48:11 2017 -0700

    Fix future schedule display timezone offset.
joey@pip:~/SIP$ git --no-pager diff
diff --git a/helpers.py b/helpers.py
index 5ea9055..f415e10 100644
--- a/helpers.py
+++ b/helpers.py
@@ -159,7 +159,7 @@ def restart(wait=1, block=False):
         except Exception:
             pass
         gv.restarted = 0
-        subprocess.Popen('service sip restart'.split())
+        subprocess.Popen('systemctl restart sip.service'.split())
     else:
         t = Thread(target=restart, args=(wait, True))
         t.start()
```

I tried fixing this instead by simply exiting (see below), relying on either openrc or systemd (or upstart) to restart the  process.  But this did not work for some reason:

```
joey@pip:~/SIP$ git --no-pager diff
diff --git a/helpers.py b/helpers.py
index 5ea9055..f415e10 100644
--- a/helpers.py
+++ b/helpers.py
@@ -159,7 +159,7 @@ def restart(wait=1, block=False):
         except Exception:
             pass
         gv.restarted = 0
-        subprocess.Popen('service sip restart'.split())
+        sys.exit(os.EX_OK)
     else:
         t = Thread(target=restart, args=(wait, True))
         t.start()
```

jharvell, 2017-07-02:
Also, I want to report a bug where the looking at the logs does not work on the Android app.  Do you know where I would report that?

jharvell, 2017-07-02:
I did not mean to close this issue.  Reopening.

Dan-in-CA, 2017-07-02:
Hi,
Did you follow the instructions in the SIP wiki for starting the program automatically with system.d?:
:https://github.com/Dan-in-CA/SIP/wiki/Installation

Look for the section called "The recommended method'.

Regarding the problem with the Android app, are you referring to the OpenSprinkler app? It does not support all of SIP's features. It is best to open SIP in a browser on your mobile device.

jharvell, 2017-07-02:
Actually, yes I did follow that.  Keep in mind everything is working fine other than when changing config that requires a restart.  This includes starting up SIP on boot.

The problem is that the restart function in helpers.py assumes there is an openrc service named sip. There is no such openrc service on my platform.

To demonstrate what happens on my box when trying to run 'service sip restart', see below.  Note I think if you changed the code as I do in my workaround it would likely work on systems that have systemd with openrc backwards compatibility, since those systemd dynamically generate systemd services based on openrc service files.

```
root@pip:/home/joey/SIP# python
Python 2.7.3 (default, Jun 21 2016, 21:00:47) 
[GCC 4.6.3] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import subprocess
>>> subprocess.Popen('service sip restart'.split())
<subprocess.Popen object at 0xb6d5c070>
>>> sip: unrecognized service

```

Dan-in-CA, 2017-07-03:
OK. I will test your work around and if it works here I will add it to the distro.

Thanks.
Dan

gdgeist, 2017-10-02:
Shouldn't this be closed? per Dan's last note?  Looking to change from Unified to SIP in order to use MQTT for my OpenHab integration (more granular information/control) and since this is a systemd install want to ensure I am not starting with an issue.

closed 2017-10-02 by Dan-in-CA