Dan-in-CA/SIP #197: Typo on Options page (minor)

jtrickey, 2017-10-15:
Under "Station Handling", the tooltip for "Individual Duration" states "Allow each station to have its own rum time in programs."

"rum" should be "run".  The sprinklers aren't allowed to party with out me.  :)

Software is great, and has really helped me out. Thank you so much!

Dan-in-CA, 2017-10-15:
jtricky,

Thanks for that. This actually comes at the perfect time.

Some users have been interested in running SIP on other hardware (Pi clones) such as orange Pi and Odroid-C2. I have decided to set up separate branches for those and keep them synced by applying updates as patches rather than merging which would combine all the code.

This minor update will give me the opportunity to  try out the patch method.

I'll let you know when the update has been added.

Dan-in-CA, 2017-10-15:
Done.
Party on! 
No Drunken sailors allowed though.

closed 2017-10-15 by Dan-in-CA

jtrickey, 2017-10-26:
Nice!  Thanks again Dan!