Dan-in-CA/SIP #198: Home page doesn't always reflect upcoming events

ndesimone, 2018-01-20:
Maybe I'm missing something really obvious.
- I created program 3 (all others disabled) with a schedule to only water the lawn for Saturday (today) and a few days next week.  Initially, the program was set to run at 7:15
- I moved the time to 14:30

[image]

[image]

- The homepage doesn't reflect the change

[image]

What am I missing?

Dan-in-CA, 2018-01-20:
Hi Nick,

I have tried duplicating your programs on my test rig and see the same problem.

It looks like it results from changing the start time when "single pass" has been set. As I recall the single pass setting is implemented by a sort of hack. It sets the stop time to be the same as the start time plus the duration to avoid multiple passes. I am not sure why it does not accept a change an I will look into the cause. It also looks like changing the setting to recurring is not saved.

I am currently working on a modification of SIPs irrigation program data format. This will involve changing how "single pass" works. I will make sure this problem is fixed.

Thanks for the bug report, I will post here when I have it fixed.

Dan

pansegrau, 2018-01-20:
Try scrolling to the following Monday.  I suspect you will not see the change until then.  I also noticed that it does not immediately update to the current day or hour.

Dan-in-CA, 2019-12-30:
I believe this has been resolved.

closed 2019-12-30 by Dan-in-CA