Dan-in-CA/SIP #200: OpenSprinkler-App interface broken

heegard, 2018-03-09:
From:  Dan-in-CA committed on Jan 8 --

The two changes to plugins/mobile_app.py of the form:
-                "fwv": gv.ver_str+'-OSPi',
+                "fwv": gv.ver_str+'-SIP',

Causes the app to reject SIP.  Either someone should update OpenSprinkler-App, or more easily, simply revert these two instances (with a note so that it is not undone later).

-Chris

Stifler6996, 2018-03-09:
I use OpenHAB2 which has an addon for OpenSprinkler. This has not been working for some time.

Whilst I do not use the mobile phone app, I suspect that this is the same reason for the OpenHAB2 module not working.

Dan-in-CA, 2018-03-09:
Hi Chris,
The change has been reverted. 
However this may be a very short term fix due to the fact that the format of SIP's irrigation programs file is being updated/modified. This change may well break the mobile app.

Have you tried accessing SIP's web UI on a mobile device lately? The addition of Bootstrap to the distro over a year ago made SIP quite usable on mobile devices and it gives you access to the plugins which the OpenSptinkler app will not support.

The changes to SIP's programs.json file (to be renamed programData.json) Make the core code much easier to read, maintain and modify.

@Stifler6996,
SIP probably will not work with OpenHAB2.
See this forum post:
http://nosack.com/sipforum/showthread.php?tid=79&highlight=OpenHAB2

heegard, 2018-03-14:
Dan,
It would be nice to try and keep SIP compatible with the OSPi app on phones.  I'm willing to work on trying to see if we can keep this working as SIP evolves.

-Chris

Dan-in-CA, 2018-03-14:
Chris,
That would be great if you would like to help keep the OSPi app/plugin working.

I have pushed a work-in-progress branch to the SIP repo named "prog_data". It is fairly complete but still needs a few kinks worked out.

There are a couple of files that should help you:
1. ProgramData_conversion_notes.txt has information abut the differences in the file formats.

2. convert_progs.py is designed to convert from the old format to the new one. It is used by SIP to do the conversion automatically but it can be run stand-alone. The old file is left in place so you can convert existing program data for testing but still revert to the older format by changing branches.

Let me know if you have any questions.

Dan-in-CA, 2019-10-26:
With the end of support for Python 2x ant the end of 2019, SIP is about to be updated to a version that will run under Python 2.7 and Python3.

There is also the revision of the file format and other changes in the new version.

The new version is now in the Py3_WIP branch. It is nearly finished and will be merged into the Master branch by the end of the year,

Anyone wishing to keep compatibility with OSPi can keep using the current version or offer a pull request for the Py3_WIP branch with a compatibility solution.

closed 2019-10-26 by Dan-in-CA