Dan-in-CA/SIP #202: Feature Request:  simplify HTTPS access to SIP

jharvell, 2018-04-30:
I have been using SIP on a Beaglebone Black for a while, running nginx on the Beaglebone black as a TLS Termination proxy.  That works fine.

I moved away from the Debian based distro on the BBB and am cross-compiling a new Gentoo distribution.  Due to various issues I am having trouble getting nginx and its dependencies cross-compiled onto this platform.  I was able to get stunnel on it, but through experimentation I discovered that stunnel cannot properly server as a TLS termination proxy for SIP due to the way it peforms redirects at login.

When SIP redirects, it redirects to http://host.from.get/login, and it sets the attribute httponly in the Set-Cookie header field.

Whatever is the TLS Termination Proxy in between has to detect this and translate the http to https in the redirect, and possibly also remove the htmlonly attribute.

Would it be possible to add a feature to SIP (a checkbox in the configuration), where when checked SIP  puts https instead of http in redirects (and does not put in the htmlonly attribute in the Set-Cookie)?

Alternatively (and preferrably), would it be possible to add HTTPS support to SIP directly?  That would really make the whole configuration much less of a chore.

Dan-in-CA, 2018-04-30:
SIP uses the web.py framework. The default is to use the included "development" web server which is an old version of cherrypy. I don't believe that can support HTTPS but there is the possibility of using lighttpd as the server and it does support HTTPS.

You can find the details in the web.py docs under CHAPTER 4 Deploying web.py applications:
https://media.readthedocs.org/pdf/webpy/latest/webpy.pdf

It requires flup to be installed:
https://www.saddi.com/software/flup/dist/

However you might want to use the more recent flup6:
https://pypi.org/project/flup6/

jharvell, 2018-04-30:
Great.  Thanks for the info.  By the way, I was able to get haproxy built for this target and get it working correctly with SIP as a TLS Termination Proxy.

mihugo, 2018-04-30:
For what it is worth. I am using nginx in front of sip as a reverse proxy for my controller. I landed on nginx over other solutions based on how small it is and how little resources it uses when compared to many of the dedicated proxy options out there. I don't recall where haproxy fell as far as resource usage.

Dan-in-CA, 2019-10-26:
No further activity on this thread.
closing. Re-open of there is still interest.

closed 2019-10-26 by Dan-in-CA