Dan-in-CA/SIP #205: water_level_adj plugin broken due to wunderground api

shred444, 2018-06-02:
Looks like wunderground has stopped their api access after IBM acquisition. Is there any chance this plugin will be ported to darksky.net? I can plan on making the modifications but it's going to take me significant amount of time to figure out what's going on here.

Dan-in-CA, 2018-06-03:
Yep. I had a feeling that would happen at some point.
Looks like darksky.net limits the free API to 1000 calls per day total for any application.
From their FAQ:
 "Can I require my users to sign up for a developer's API key?
    No, your end-users should not sign up for Dark Sky API keys as a means for you to circumvent the daily cap of 1,000 free calls. The 1,000 free calls per day are intended for one developer or company to explore the API."

OpenWeatherMap. may be a better option. See this issue on the SPI Plugins repo:
https://github.com/Dan-in-CA/sip_plugins/issues/31

If you would be willing to try making the modifications that would be great. I would be glad to provide advice.

I am working on on a couple of other urgent projects otherwise I would give it a go.

Dan

mw46d, 2018-07-16:
[image]
It looks like my free API key is still working? What did I right (or wrong)??

mw46d, 2018-07-16:
https://openweathermap.org/api does not seem to provide historical data for the `free` accounts:-( The current plugin uses previous three days + current day forecast  + next three days forecast :-( Switching away from the historical data would impact the accuracy:-(

DavidLaidlaw, 2018-07-16:
Yes, it would require keeping history.   After three days of running, it would have as much history as is currently used. For the first three days it would be less accurate.

WU has better coverage, but if it is not accessible to new users, that seems very limiting.  It does seem to have a grandfathering or existing keys, at least for now.  I don’t have one, unfortunately, so am enthusiastic about an alternative!

David

David Laidlaw, Brown CS
Box 1910, Prov., RI 02912
+1.401.354.2819

DavidLaidlaw, 2018-07-16:
> There is a difference between historical forecast and actual measured data.:-(

True, but the actual current weather is available in addition to the forecast, and the current weather can be saved to use as history.  

D

David Laidlaw, Brown CS
Box 1910, Prov., RI 02912
+1.401.354.2819

Dan-in-CA, 2019-12-30:
The weather_level_adj plugin is working with OpenWeatherMap.

closed 2019-12-30 by Dan-in-CA