Dan-in-CA/SIP #206: Enabling Individual Duration Setting Doesn't Persist

Tails86, 2018-06-04:
I have noticed that the Individual Duration setting resets to false at some point over time. All of my other settings persist, but the one doesn't want to stay. It has happened twice over a year. Simply doing reboot, reset, or system update doesn't cause it. I don't have any other clues as to what it could be.

Dan-in-CA, 2018-06-04:
I'll take a look but that sounds like a hard one to track down if it only happens a couple of time a year.

Dan-in-CA, 2018-06-04:
Are you using any plugins?
That might give a clue.

Dan-in-CA, 2018-06-04:
The OpenSprinkler app has diverged from SIP. The individual duration settings in OpenSprinkler and SIP are different.
There is probably not going to be a fix for this problem.
Have you tried using SIP's web UI on your android device lately? It is actually pretty usable and you can access plugin settings which the OpenSprinkler app can't support.

Tails86, 2018-06-04:
Fair enough. The only reason I use the OpenSprinkler app is because I can more quickly get in without having to enter the IP then password (all that is saved, so fewer taps). I usually only use it to manually start a station which seems to work fine for now.

Dan-in-CA, 2018-06-04:
I made a browser shortcut on my phone (add to home screen). It only takes one tap to open.

Tails86, 2018-06-06:
Yeah, you're right. That is easy enough to do that and have chrome save my password. Then I can avoid incompatibilities like this.

closed 2019-10-26 by Dan-in-CA