Dan-in-CA/SIP #207: Weather Adjustment: Location issue

Dalmapalma, 2018-06-10:
Hi,

I just installed the weather ajdustment plugin. I added the city name 'Steinen'of my hometown as the ZIP Code or the coordinates were not working. But in Germany there are many cities called Steinen...How can I be sure that it is taken the right one?

Dan-in-CA, 2018-06-11:
The weather plugins are not working properly due to changes weather underground.
See this issue:
https://github.com/Dan-in-CA/SIP/issues/205
Hopefully we will have an update before too long.

Dan-in-CA, 2019-10-26:
The weather_level_adj plugin has been updated to use OpenWeathermap and location is by Latitude - Longitude (GPS) coordinates. That should avoid city name conflicts.

closed 2019-10-26 by Dan-in-CA