Dan-in-CA/SIP #208: Not working with my relay board

schmatzp, 2018-06-20:
Hi Dan,
when trying your code, your sensor 1 correspond to my relay 3 (I only need 3 relays connected to 11, 12 and 13 based on GPIO.BOARD addressing)
I've tested a simple python code, and it is working, means my wiring is correct.
How can I modify your code to correspond to my need.

PS: this is a pi 2

Thanks.
Pascal

Dan-in-CA, 2018-06-20:
Hi Pascal,
I'm not sure what you are referring to as "saensor 1".
Have you installed the relay_board plugin?
Instructions are here:
https://github.com/KanyonKris/relay_board/wiki
In the instructions OSPi = SIP

Dan-in-CA, 2018-06-27:
It has been a week. I guess the problem is solved so I am closing the issue.
You can always reopen it if you still haven't figured out the problem.

closed 2018-06-27 by Dan-in-CA