Dan-in-CA/SIP #209: Question: sessions/* files created & deleted without using the web interface?

mw46d, 2018-07-15:
Hello,
it looks like something is causing a constant stream of files to be created in the sessions/ directory?! I'm seldom need to access the web frontend, but those show still up?

Any idea?

mw46d, 2018-07-15:
I just pointed the code to a tmpfs directory for now and restarted the service. But without any [human] accesses to the web frontend, I get

pi@ospi ~/OSPi $ ls -ltr /run/OSPi/sessions/
total 60
-rw-r--r-- 1 root root 167 Jul 15 14:01 aa6f14f2da6d631b4f0ad1f0438d05cf10242a0a
-rw-r--r-- 1 root root 167 Jul 15 14:01 6fad8c0847bc6ad73bcf24acb5e484ee04bed90a
-rw-r--r-- 1 root root 167 Jul 15 14:01 e57682db27bfd4348d6e578f31880764daa8005b
-rw-r--r-- 1 root root 167 Jul 15 14:01 58ee4542208e8c16c321e2d9762ea4d19557c558
-rw-r--r-- 1 root root 167 Jul 15 14:01 ae57002fc175ae535aeaa2bcd4d6316f3a80bf44
-rw-r--r-- 1 root root 167 Jul 15 14:02 c4876a3ddb7a37519584f691b35749ad8077dbca
-rw-r--r-- 1 root root 167 Jul 15 14:02 1c2209bfe6aeb9d8b408f6de8f0b8e79f17e54ae
-rw-r--r-- 1 root root 167 Jul 15 14:02 35fc1811b9c33cecd258809843fc305494251f53
-rw-r--r-- 1 root root 167 Jul 15 14:02 ad8fb46b18ff6d163d962dff3e95cde1ef9b22b0
-rw-r--r-- 1 root root 167 Jul 15 14:02 73e7cf2d6d996ca9df451b3290d1141b02fe922d
-rw-r--r-- 1 root root 167 Jul 15 14:03 53d62806291faac5eebf6e06a4368fc493b02400
-rw-r--r-- 1 root root 167 Jul 15 14:03 d5c92ea19a56ecebf4be83d996d2a2e6eda3911f
-rw-r--r-- 1 root root 167 Jul 15 14:03 eaddd876cd3d73c89d3e0197ab0d166f97198ad1
-rw-r--r-- 1 root root 167 Jul 15 14:03 e641bbafff2e3add42fa1e8659cbbda97c992458
-rw-r--r-- 1 root root 167 Jul 15 14:03 ec0d0f3301b0aa2a2089de01d315488757d5d7ba

mw46d, 2018-07-15:
```diff
pi@ospi ~/OSPi $ git diff 
diff --git a/sip.py b/sip.py
index 50ec976..02c3ba4 100755
--- a/sip.py
+++ b/sip.py
@@ -206,7 +206,7 @@ app = SIPApp(urls, globals())
 #  disableShiftRegisterOutput()
 web.config.debug = False  # Improves page load speed
 if web.config.get('_session') is None:
-    web.config._session = web.session.Session(app, web.session.DiskStore('sessions'),
+    web.config._session = web.session.Session(app, web.session.DiskStore('/run/OSPi/sessions'),
                                               initializer={'user': 'anonymous'})
 template_globals = {
     'gv': gv,
diff --git a/sip.service b/sip.service
index 99d469c..e567956 100644
--- a/sip.service
+++ b/sip.service
@@ -4,6 +4,7 @@
 Description=SIP
 After=syslog.target network.target
 [Service]
+ExecStartPre=mkdir -p /run/OSPi/sessions
 ExecStart=/usr/bin/python /home/pi/SIP/sip.py
 Restart=on-abort
 WorkingDirectory=/home/pi/SIP/
```

mw46d, 2018-07-16:
Something is clearly going on. This is a rather old system upgraded to your latest released version. Not sure, what's left around here?!
```
root@ospi:/home/pi/OSPi# /usr/bin/python ospi.py
ALARM from Example_Sender!: Just an example!
plugins loaded:
  lcdd_adj
  mobile_app
  plugin_manager
  signaling_examples
  system_update
  weather_level_adj
zones changed
[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
 Starting timing loop

http://0.0.0.0:8080/
Checking weather status...
LCDd plugin is active
Using 7 days of information.
Water needed (7 days): 67.3mm
Total rainfall       : 0.0mm
_______________________________-
Irrigation needed    : 67.3mm
Weather Adjustment   : 200.0%
127.0.0.1:46914 - - [15/Jul/2018 18:49:02] "HTTP/1.1 GET /" - 303 See Other
127.0.0.1:46915 - - [15/Jul/2018 18:49:04] "HTTP/1.1 GET /login" - 200 OK
127.0.0.1:46916 - - [15/Jul/2018 18:49:04] "HTTP/1.1 GET /sn0" - 303 See Other
127.0.0.1:46917 - - [15/Jul/2018 18:49:05] "HTTP/1.1 GET /" - 303 See Other
127.0.0.1:46918 - - [15/Jul/2018 18:49:05] "HTTP/1.1 GET /login" - 200 OK
Some Stations has been scheduled: [[0, 0, 0, 0], [0, 0, 0, 0], [0, 0, 0, 0], [0, 0, 0, 0], [1531680600, 1531681680, 1080, 2], [0, 0, 0, 0], [0, 0, 0, 0], [1531681680, 1531682760, 1080, 2], [1531682760, 1531683840, 1080, 2], [0, 0, 0, 0], [0, 0, 0, 0], [0, 0, 0, 0], [0, 0, 0, 0], [0, 0, 0, 0], [0, 0, 0, 0], [0, 0, 0, 0]]
zones changed
[0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
127.0.0.1:46919 - - [15/Jul/2018 18:50:01] "HTTP/1.1 GET /" - 303 See Other
127.0.0.1:46920 - - [15/Jul/2018 18:50:01] "HTTP/1.1 GET /login" - 200 OK
127.0.0.1:46921 - - [15/Jul/2018 18:50:01] "HTTP/1.1 GET /sn0" - 303 See Other
127.0.0.1:46922 - - [15/Jul/2018 18:50:02] "HTTP/1.1 GET /" - 303 See Other
127.0.0.1:46923 - - [15/Jul/2018 18:50:02] "HTTP/1.1 GET /login" - 200 OK
127.0.0.1:46924 - - [15/Jul/2018 18:51:01] "HTTP/1.1 GET /" - 303 See Other
127.0.0.1:46925 - - [15/Jul/2018 18:51:01] "HTTP/1.1 GET /login" - 200 OK
127.0.0.1:46926 - - [15/Jul/2018 18:51:01] "HTTP/1.1 GET /sn0" - 303 See Other
127.0.0.1:46927 - - [15/Jul/2018 18:51:01] "HTTP/1.1 GET /" - 303 See Other
127.0.0.1:46928 - - [15/Jul/2018 18:51:02] "HTTP/1.1 GET /login" - 200 OK
127.0.0.1:46929 - - [15/Jul/2018 18:52:01] "HTTP/1.1 GET /" - 303 See Other
127.0.0.1:46930 - - [15/Jul/2018 18:52:01] "HTTP/1.1 GET /login" - 200 OK
127.0.0.1:46931 - - [15/Jul/2018 18:52:01] "HTTP/1.1 GET /sn0" - 303 See Other
127.0.0.1:46932 - - [15/Jul/2018 18:52:01] "HTTP/1.1 GET /" - 303 See Other
127.0.0.1:46933 - - [15/Jul/2018 18:52:02] "HTTP/1.1 GET /login" - 200 OK
127.0.0.1:46934 - - [15/Jul/2018 18:53:01] "HTTP/1.1 GET /" - 303 See Other
127.0.0.1:46935 - - [15/Jul/2018 18:53:01] "HTTP/1.1 GET /login" - 200 OK
127.0.0.1:46936 - - [15/Jul/2018 18:53:01] "HTTP/1.1 GET /sn0" - 303 See Other
127.0.0.1:46937 - - [15/Jul/2018 18:53:01] "HTTP/1.1 GET /" - 303 See Other
127.0.0.1:46938 - - [15/Jul/2018 18:53:02] "HTTP/1.1 GET /login" - 200 OK
```

mw46d, 2018-07-16:
OK, I think, I found the cause;-)
```
root@ospi:/var/spool/cron/crontabs# ls -l www-data 
-rw------- 1 www-data crontab 276 Apr 26  2014 www-data
root@ospi:/var/spool/cron/crontabs# cat www-data 
# DO NOT EDIT THIS FILE - edit the master and reinstall.
# (/tmp/crontab.txt installed on Sat Apr 26 19:19:42 2014)
# (Cron version -- $Id: crontab.c,v 2.13 1994/01/17 03:20:37 vixie Exp $)
* * * * * cd /var/www/sprinklers; php /var/www/sprinklers/watcher.php >/dev/null 2>&1
root@ospi:/var/spool/cron/crontabs# crontab -e -u www-data
crontab: installing new crontab
root@ospi:/var/spool/cron/crontabs# ls -l www-data 
-rw------- 1 www-data crontab 289 Jul 15 19:11 www-data
```
It looks like the apache on :80 forwards those requests to :8080

Dan-in-CA, 2018-07-16:
Yes, all those /login lines are strange.
Are you running the OpenSprinkler app or some other way of accessing the program?
Those /sn0 lines look like something is trying to read or write a station status using the HTML GET commands.

By the way the weather plugin may not be working anymore. See this issue:
https://github.com/Dan-in-CA/SIP/issues/205

You might try disabling some or all plugins and see if that help.

mw46d, 2018-07-16:
The home page still says:
`Plugin adjustments: 200%`, not sure, iff that's just the old value?! But for now, that's the right value for now anyhow. No rain or cooler weather in the sight;-)

mw46d, 2018-07-16:
OK, this was a left-over checking setup from the sprinklers web-app. Nothing to do with SIP

closed 2018-07-16 by mw46d