Dan-in-CA/SIP #211: Question: is the serial lines (pin 8, 10) available for use by 3rd programs ?

FulvioSpelta, 2018-10-21:
Hi all, i'd like to use SIP to control my little garden. It has only 2 valve so my idea is to use SIP with 2 control rele directly conntected to gpio pins. Due to limted dimension of the garden i'd like to use the raspberry for some other little activites where one of them needs to use the serial line.

Reading forum and docs have undrtood that the RX/TX pins are used, at least the 8 pin, by SIP (8 for rian sensor).

Is it possibile to free the rx/tx lines moving rain sensor, etc to others pins or, at least, don't use them ?
Thanks for the support.

Dan-in-CA, 2018-10-21:
Hi Fulvio,
SIP will only use the rain sensor pin if you tick the box in OPTIONS > Rain Sensor > Use rain sensor.
If you do not tell SIP to use a rain sensor it will not try to use that pin.

You can also change the pin SIP uses tor the rain sensor by editing the file "gpio_pins.py" line 61.

If you will be connecting your relays directly to the Pi I recommend using the  relay_16 plugin.

FulvioSpelta, 2018-10-22:
Hi Dan,
thanks for the hints. 
Before to read your post I've modified the "gpio_pins.py" in pin mapping line 20 (i'm using an old raspi) as follow:
```python
gv.pin_map = [0,0,0,0,0,5,0,7,11,0,19,13,15,0,0,0,0,0,0,0,0,0,0,0,0,0,0]
```

I've enabled the rain sensor (i'd like to use it) and currently both programs (SIP and the program that use the serial line) are running properly.

I'm using "relay_board" plugin, do you think that i've to change it ?

Jst a note: frome code at line 55 - 78 it seems that the rain and relay pins are always configured (maybe i'm wrong i've not analyzed all code :-) )

```python
global pin_rain_sense
global pin_relay

try:
    if gv.platform == 'pi':  # If this will run on Raspberry Pi:
        GPIO.setmode(GPIO.BOARD)
        pin_rain_sense = gv.pin_map[8]
        pin_relay = gv.pin_map[10]
    elif gv.platform == 'bo':  # If this will run on Beagle Bone Black:
        pin_rain_sense = gv.pin_map[15]
        pin_relay = gv.pin_map[16]
except AttributeError:
    pass

try:
    if gv.use_pigpio:
        pi.set_mode(pin_rain_sense, pigpio.INPUT)
        pi.set_pull_up_down(pin_rain_sense, pigpio.PUD_UP)
        pi.set_mode(pin_relay, pigpio.OUTPUT)
    else:      
        GPIO.setup(pin_rain_sense, GPIO.IN, pull_up_down = GPIO.PUD_UP)
        GPIO.setup(pin_relay, GPIO.OUT)
except NameError:
    pass
```

Dan-in-CA, 2018-10-22:
If the relay_board plugin is working for you keep it.

Pin 10 is configured to work with an OpenSprinkler board which has a small on-board relay. If you are not using that board and the on-board relay, pin 10 is free to use for other purposes. The on-board relay is only activated by a plugin ("relay") and not used by SIP directly.

You could probably comment out the "pin_relay" lines in the code without affecting the program's operation.

FulvioSpelta, 2018-10-28:
Thanks for the help and the work.

closed 2018-10-28 by FulvioSpelta