Dan-in-CA/SIP #213: Main webpage "calendar buttons" broken for Firefox

Stifler6996, 2018-12-02:
Hi,
When on the homepage, in Firefox the Calendar program (and history) do no display. If you press NextDay you will get the next day displayed.

From there, if you press PreviousDay it will go back, however you can not go back to "Today".

If you press NextDay (any number of times) and then press the Today button, it does not return to the Today page. (Just a guess but it looks like all it might be going to the "non-displayed Today" page and programs not being displayed)

The PrevDay button doent appear to work on the "Today" page in any situation.

Dan-in-CA, 2018-12-03:
Thanks for reporting that. I see it also.
I'll let you know as soon as I find and fix the problem.

Dan

Dan-in-CA, 2018-12-03:
Nope. Booth copies are working as expected.
I am running firefox V 63.0.3 (64-bit)

@Stifler6996  Do you have logging enabled (Options > Logging)?
There needs to be some log data for the day you are looking at for the run times to show on a previous day.  If you open the Log from the menu at the top of the page you should see a list of run data. If not the log is missing or not enabled.

You also need to have some irrigation scheduled for a particular day in the future to see the run times for that day.

Dan-in-CA, 2018-12-04:
Thank you for the feedback.
That is good to know even though we will probably never know the details.

Dan

closed 2018-12-09 by Stifler6996