Dan-in-CA/SIP #218: Plan for Python 3?

heegard, 2019-02-20:
Dan,

As Python 2.7 is to be phased out at the end of the year, shouldn't we make a plan to migrate to python 3?  I'm more than willing to help...

-Chris

Dan-in-CA, 2019-02-20:
Yes definitely.

In fact I have already started working on it. 
One major concern was the fact the SIP is built on web.py which was a Python 2x framework. Fortunately there has been renewed activity on web.py development and there is a (experimental) Python 3 compatible version available.

I have been using the 0.40-dev1 version of web.py on my test system with no problems.
In order to keep SIP installation as simple as possible (just download and run), I have included the new web.py files and dependencies in the SIP directory. These include: 

Cheroot-6.4.0
backports.functools-lru-cache-1.5 
more-itertools-4.3.0 
six-1.11.0

Plus the new web folder from the web.py GitHub repo:
https://github.com/webpy/webpy

I welcome your offer to help with the update. I have posted a development branch with the Python 3 compatible web.py included and we can go from there. The branch is named P3_development. It also includes the latest update of the modified program date changes.

By the way, the version of SIP with the new Cheroot server has **much** quicker page loads.
 
My time is a bit limited right now because I have been working on a stand-alone (no internet) solar powered wifi enabled version of sip for some friends.  It was working great until the last rain storm when the "weatherproof" enclosure got full of water...

Dan-in-CA, 2019-10-26:
A Python3 compatible version will be available by the end of the year.
It is currently the Py3_WIP branch. It is working very well (pages load much faster) and the plugins are currently being updated.

closed 2019-10-26 by Dan-in-CA