Dan-in-CA/SIP #219: Weather Adjustment

constantly-lost, 2019-03-21:
Since Weather Underground is no longer free have you thought about using Dark Sky API. Free for 1000 calls per day. Can your plugin be modified to accept data from them? Your SIP is great by the way. Thanks

Dan-in-CA, 2019-03-24:
There has been some discussion of an alternative weather source.
See https://github.com/Dan-in-CA/SIP/issues/205

I don't know of anyone working on an update yet.

There is also some information on the SIP forum. 
https://nosack.com/sipforum/showthread.php?tid=141&pid=648#pid648

One suggestion is to use 7timer.info. It may be a useful solution.

airsoftsoldrecn9, 2019-05-09:
Would like to create a PR for updating the plugins on this project. 
I replaced Wunderground with OpenWeatherMaps API. Dark Sky is an option, just need to test with their API, and provide the user the option to switch keys and APIs in templates. 
[image]

Dan-in-CA, 2019-05-09:
Looks good.

The PR would need to be on the SIP plugins repo:
https://github.com/Dan-in-CA/sip_plugins

fennsen64, 2019-05-14:
Hi, I've taken a similar approach, but migrated the existing plugin to the WU API (of course you need to be owner of a PWS to get a new API key. 
So I can still use my own history data which is pushed from my PWS (based on weewx) to wunderground.

On top I included a (optional) MQTT functionallity which pushes the history & forecast data to a broker (which gives the opportunity to use this data outside from SIP (e.g. pimatic, ioBroker).
In addition this python script can be used outside from the SIP machine - standalone. In this case it pushes the final weather-adjsutment via HTTP SIP API call.
I didn't yet adjust/update the config-webpage, so configuration is made within the script.

As this code is very long, where to put it? I've an GIT account but never reated a repo by myself...

Dan-in-CA, 2019-05-14:
That sounds very interesting.
The best place to put the files is in an open source GitHub repository.
The instructions are pretty easy:
https://help.github.com/en/articles/create-a-repo

There is also a video on youtube:
https://www.youtube.com/watch?v=LR5BYZjuXMU

FulvioSpelta, 2019-05-21:
@airsoftsoldrecn9  can you kindly share the code in the meanwhile of the PR? Thanks a lot

fennsen64, 2019-05-22:
Hi @ all.
Pls find my above described implementation based on the new WU API (V3) here:
https://github.com/fennsen64/OSPI-wunderground-adjust
Running stable in my environment but happy for more testers. have fun!

airsoftsoldrecn9, 2019-05-22:
Sorry for the extreme delay. Please review https://github.com/Dan-in-CA/sip_plugins/pull/33. I welcome feedback for improvement.

Dan-in-CA, 2019-10-26:
The weather_level_adj plugin has been updated further.

Re-open of there are still any issues.

closed 2019-10-26 by Dan-in-CA