Dan-in-CA/SIP #220: Application Starts with IP .0

konoichi, 2019-05-02:


konoichi, 2019-05-02:
I did a fresh install on a new rasbian stretch light distro and ran into an issue after starting the application. It seems that SIP is not able to resolve the local IP address as SIP URL/IP is shown as 0.0.0.0:80.
I needed to edit the sd.json file and set the IP manually and I changed the port to 8080.
Now the application is running as it should but I hope someone can explain what whent wrong here.

Dan-in-CA, 2019-05-02:
Not sure why you are seeing this.
There was a recent change to make the default IP address compatible with both IPv4 and IPv6. as described in issue #216:
https://github.com/Dan-in-CA/SIP/issues/216

You should be seeing something like http://:::80/ which is the new default.

closed 2019-10-26 by Dan-in-CA

hemangjoshi37a, 2024-06-18:
how to change the default port 80 to anything else ?

Dan-in-CA, 2024-06-18:
As the original poster mentioned you can edit the file SIP/data/sd.json. look for the line ""htp": 80, and chant the port there.