Dan-in-CA/SIP #221: Port forwarding

WoutVanH, 2019-05-14:
Hello everyone,

First I want to thank the creator of this great program! It has a lot of possibles that I doesn't find by Hunter, Rainbird or another big company... 

If you would like dutch as optional language I would like to translate this. This is the least I can do back for you ...

I am a new user and am very happy with the program! 
I wish to port forward the webinterface, is there any documentation about it? Is it possible? 

Thanks for a response!

Regards,
Wout (Belgium)

WoutVanH, 2019-05-14:
Hello,

I just get portforwarded my controller... But I have another problem, I set the security options on but the password I chosen isn't correct... How can I choose a new password without my old?

Regards,
Wout

WoutVanH, 2019-05-14:
Hello, I fixed already. I am running another few webistes and these give a some issues with each other.

Dan-in-CA, 2019-05-14:
Welcome Wout,
A Dutch translation of the user interface would be great!
Instructions for doing the translation are on the SIP Wiki:
https://github.com/Dan-in-CA/SIP/wiki/Translation-doc

An alternative to port forwarding for secure remote access to SIP is to use a VPN. I have used pivpn with good success.
http://www.pivpn.io/

closed 2019-10-26 by Dan-in-CA