Dan-in-CA/SIP #222: internal server error  500

aocana, 2019-05-17:
hi there,

when i try to run manually the program, executing program "run once" i'm getting error  "500 internal server error"

here the log:

`tazo@-sip:~/SIP$ sudo /usr/bin/python  /home/tazo/SIP/sip.py
ALARM from Example_Sender!: Just an example!
Monthly Adjust: Setting water level to 100%
Pressure plugin is active
Waiting...
telegramBot plugin is active
Update "None" caused error "Conflict: terminated by other getUpdates request; make sure that only one bot instance is running (409)"
plugins loaded:
  mobile_app
  monthly_adj
  mqtt
  mqtt_schedule
  mqtt_slave
  mqtt_zones
  plugin_manager
  pressure_adj
  relay
  relay_board
  signaling_examples
  system_update
  telegramBot
  weather_adj
  weather_level_adj
Starting timing loop

http://:::80/
Update "None" caused error "Conflict: terminated by other getUpdates request; make sure that only one bot instance is running (409)"
Update "None" caused error "Conflict: terminated by other getUpdates request; make sure that only one bot instance is running (409)"
::ffff:192.168.1.214:53474 - - [17/May/2019 14:19:02] "HTTP/1.1 GET /" - 200 OK
::ffff:192.168.1.214:53474 - - [17/May/2019 14:19:02] "HTTP/1.1 GET /static/images/favicon-16x16.png" - 200
::ffff:192.168.1.214:53488 - - [17/May/2019 14:19:03] "HTTP/1.1 GET /api/status" - 200 OK
::ffff:192.168.1.214:53486 - - [17/May/2019 14:19:03] "HTTP/1.1 GET /api/log" - 200 OK
::ffff:192.168.1.214:53488 - - [17/May/2019 14:19:04] "HTTP/1.1 GET /api/log" - 200 OK
Update "None" caused error "Conflict: terminated by other getUpdates request; make sure that only one bot instance is running (409)"
::ffff:192.168.1.214:53488 - - [17/May/2019 14:19:11] "HTTP/1.1 GET /vr" - 200 OK
::ffff:192.168.1.214:53488 - - [17/May/2019 14:19:11] "HTTP/1.1 GET /api/log" - 200 OK
Some Stations has been scheduled: [[0, 0, 0, 0], [0, 0, 0, 0], [1558102756, 1558102816, 60, 98], [0, 0, 0, 0], [0, 0, 0, 0], [0, 0, 0, 0], [0, 0, 0, 0], [0, 0, 0, 0]]
Traceback (most recent call last):
  File "/home/takezo/SIP/web/application.py", line 239, in process
    return self.handle()
  File "/home/takezo/SIP/web/application.py", line 230, in handle
    return self._delegate(fn, self.fvars, args)
  File "/home/takezo/SIP/web/application.py", line 423, in _delegate
    return handle_class(cls)
  File "/home/takezo/SIP/web/application.py", line 399, in handle_class
    return tocall(*args)
  File "/home/takezo/SIP/webpages.py", line 386, in GET
    schedule_stations(stations)
  File "/home/takezo/SIP/helpers.py", line 448, in schedule_stations
    report_stations_scheduled()
  File "/home/takezo/SIP/helpers.py", line 61, in report_stations_scheduled
    stations_scheduled.send('SIP',txt=txt)
  File "/home/takezo/SIP/blinker/base.py", line 267, in send
    for receiver in self.receivers_for(sender)]
  File "/home/takezo/SIP/plugins/telegramBot.py", line 227, in notifyStationScheduled
    txt += get_running_programs_rs()
  File "/home/takezo/SIP/plugins/telegramBot.py", line 253, in get_running_programs_rs
    txt += '\n<b>{}</b> - '.format(sname)
UnicodeEncodeError: 'ascii' codec can't encode character u'\xed' in position 15: ordinal not in range(128)

::ffff:192.168.1.214:53488 - - [17/May/2019 14:19:19] "HTTP/1.1 GET /cr" - 500 Internal Server Error
::ffff:192.168.1.214:53488 - - [17/May/2019 14:19:20] "HTTP/1.1 GET /favicon.ico" - 303 See Other
Update "None" caused error "Conflict: terminated by other getUpdates request; make sure that only one bot instance is running (409)"
Update "None" caused error "Conflict: terminated by other getUpdates request; make sure that only one bot instance is running (409)"
::ffff:192.168.1.214:53522 - - [17/May/2019 14:20:17] "HTTP/1.1 GET /api/log" - 200 OK
Update "None" caused error "Conflict: terminated by other getUpdates request; make sure that only one bot instance is running (409)"
Update "None" caused error "Conflict: terminated by other getUpdates request; make sure that only one bot instance is running (409)"
::ffff:192.168.1.214:53542 - - [17/May/2019 14:21:19] "HTTP/1.1 GET /api/log" - 200 OK`

any idea?

Dan-in-CA, 2019-05-17:
Looks like one of the plugins is causing the problem.
The 
telegramBot
weather_adj
weather_level_adj
and pressure_adj
plugins would be my guess for causing problems.

I would disable at least those and see if that eliminates the problem. Then enable them one at a time to find out which is causing the problem.

Those plugins were contributed by users and have not been updated recently.

The weather plugins are known to not be working at this time.

airsoftsoldrecn9, 2019-06-03:
Will look into this issue; quite certain my updates to at least 2 of those plugins are suspect.

aocana, 2019-06-03:
hi Dan, 

you were right! i disabled those plugings and now system works without problem. 
At this time I need to take a time to enable one by one and find out which is causing the problem. 

I will let you know soon!

thank you so much for your support!

aocana, 2019-07-19:
Hi Dan, 

not sure why , because SIP was working perfectly since last time we chated, but starting today If i try to check into SIP menu program i'm getting again error 500. here you have the log:

`tazo@alcortwo-sip:~/SIP$ sudo /usr/bin/python /home/tazo/SIP/sip.py
plugins loaded:
  mobile_app
  plugin_manager
  relay
  system_update
Starting timing loop

http://:::80/
Traceback (most recent call last):
  File "/home/tazo/SIP/web/application.py", line 239, in process
    return self.handle()
  File "/home/tazo/SIP/web/application.py", line 230, in handle
    return self._delegate(fn, self.fvars, args)
  File "/home/tazo/SIP/web/application.py", line 423, in _delegate
    return handle_class(cls)
  File "/home/tazo/SIP/web/application.py", line 399, in handle_class
    return tocall(*args)
  File "/home/tazo/SIP/webpages.py", line 394, in GET
    return template_render.programs()
  File "/home/tazo/SIP/web/template.py", line 1020, in template
    return self._base(t(*a, **kw))
  File "/home/tazo/SIP/web/template.py", line 881, in __call__
    return BaseTemplate.__call__(self, *a, **kw)
  File "/home/tazo/SIP/web/template.py", line 808, in __call__
    return self.t(*a, **kw)
  File "templates/programs.html", line 145, in __template__
    $if gv.sd['idd'] :
TypeError: 'int' object has no attribute '__getitem__'

::ffff:192.168.1.217:55184 - - [19/Jul/2019 09:35:44] "HTTP/1.1 GET /vp" - 500 Internal Server Error`

any idea ?

Dan-in-CA, 2019-07-19:
The error means that when you try to open the program page SIP is looking for a list of numbers but is only finding a single number.

I will try to duplicate this and look for a solution.

aocana, 2019-07-23:
THANKS @Dan-in-CA for your support!
I will wait for your solution. is there any plan in your mind to release this fix..?

thanks!!, thanks!!

Dan-in-CA, 2019-08-03:
I have not been able to produce the error you are getting.
Without getting the error I can't fix it. everything seems to be working fine.

Is there anything you changed in SIP just before the error started?

aocana, 2019-08-09:
Hi Dan, 

finally i have installed from scratch.
:-/

closed 2019-10-26 by Dan-in-CA