Dan-in-CA/SIP #223: Where does SIP get its time?

ndesimone, 2019-07-16:
Pi system time is correct (in console), but SIP display is off.  SIP underlying schedule is correct.

[image]

Dan-in-CA, 2019-07-16:
SIP gets it's time from the internet and is adjusted by the time zone setting on the Pi.
The idea is that if someone is accessing SIP from a remote location (in a different timezone) the display will show the time at the location of the SIP installation rather that the time on the display device.

ndesimone, 2019-07-16:
Thanks for the quick response Dan.

The machine I'm accessing the Pi is on the same wired network.  Looking at the screenshot:
- The upper right shows the viewing system time as 21:28 PDT
- The terminal window is the Pi running SIP, it's showing 21:28 PDT as well
- SIP is showing 20:28 on both the clock and the grid

HOWEVER, noticed that the program was running (3rd station is running - as it should have been based on the program).

PDT transition happened a while ago, do you know if SIP needs to be restarted after that to get it to realize that DST has happened?

Dan-in-CA, 2019-07-16:
Yes. It seems to me that I have had to restart the Pi to get it to update the time after a change to DST.

However there is a way to have the Pi get time updates using the NTP protocol and program. It is usually described for use with a real time clock but can be used without one. It will then adjust for time changes automatically.
See this post on the Raspi blog:
http://raspberrypi.tomasgreno.cz/ntp-client-and-server.html

ndesimone, 2019-07-28:
Hey Dan,
So I double checked everything I could think of:
- The PI is running NTP and sync'd with time servers

`pi@ospi ~ $ ntpq -p
     remote           refid      st t when poll reach   delay   offset  jitter
==============================================================================
+lax1.m-d.net    204.9.54.119     2 u  471 1024  377   30.120    2.382 129.915
*clock.xmission. .XMIS.           1 u  741 1024  377  102.681  -26.419 114.858
 192.138.210.214 129.7.1.66       2 u  11d 1024    0  152.440  -26.292   0.000
+ntp2.la1.hostbr 216.218.254.202  2 u  682 1024  377   37.230    6.213 195.486`

- The current time on the PI correct.  If I run `date` it shows the correct time and DST state
- The programs start and stop at the correct times

The time on the SIP home page was still off by an hour, so I simply did a `sudo service sip restart` and now the time and grid display are correct.  As we suspected, the easiest solution is to bounce SIP after a DST transition.  

Just to state clearly for others, the actual operations of SIP is correct (programs start at the right time).  This issue is only affecting the time display and grid display on the home page.

Dan-in-CA, 2019-10-26:
re-open if there is further input.

closed 2019-10-26 by Dan-in-CA

MartinMMx, 2020-11-05:
I sometimes have the issue that the OS doesn't get NTP time on bootup (e.g. because there was a powercut and the DSL takes longer to come up than the Pi), then the Pi OS will use the default time (no RTC), and never sync again with NTP. So it happens to be by *months* or *years* off.
Now the problem with that: it's not visible on the SIP UI that this issue is present, as the UI gets its time not from the Pi but from somewhere else. But the programs will run by system time, and so happen at a completely wrong time or day.

So in my opinion, the UI should at least show a big red warning that the system time has a mismatch.

Dan-in-CA, 2020-11-05:
Because SIP has a web based user interface the time displayed on the web pages comes from the device running the web browser.

The code is supposed to show the time from the Pi in the UI but that does not seem to be working properly. The idea is to show the time at the Pi's location of it is being accessed from a remote location perhaps in a different time zone. It has been tricky to test.
I will look for a solution.

MartinMMx, 2021-01-28:
Fortunately, in Germany we do not have frequent power cuts, otherwise it would be a frequent problem ;)
And well, I am not running any PiOS but this is a BeagleBoneBlack, but that's a detail that should not matter much.

The tricky part is that the NTP server allows SYNC only *much* later than the DHCP assigns the address, so you have to implement some retry-scheme for the initial SYNC. I have done this in the past for an embedded device, but there I had the control over the whole OS. Here, I would not want to differ from the default OS config much as it increases maintenance effort in case of updates.

Other than that, this is not primarily about me (as an Embedded Software Engineer) needing a solution, but rather about improving the system for all users :)