Dan-in-CA/SIP #224: TCP remote Water flow/pressure/temperature meter

catkiller007, 2019-07-29:
I'm planning to develop a remote unit based on Arduino and Ethernet communication with SIP.

I believe a modular approach will be better because not all system can be in the same place all the time.

A similar idea was used here with a remote relay board:
https://blog.danman.eu/wifi-enabled-sprinkler-controller/

I can do all the Arduino firmware and hardware but i will need some help on the plugin side.

Adrian.

Dan-in-CA, 2019-07-29:
Hi Adrian,
Agreed that a modular approach is the best way to add sensing of various types to SIP.

I will be glad to help with the plugin.
There are some things you might want to check out before hand such as the HTTP GET commands that can be used to control sip:
https://github.com/Dan-in-CA/SIP/wiki/SIP_GET_commands.pdf

There are also some MQTT plugins that may prove useful.

And check out the cli-control plugin that allows using command line commands to communicate with SIP.

I have also developed some Arduino modules that use MODBUS over RS-485 serial but have not yet made a plugin for it or published the code. That of course requires a wired connection although I have been using it with standard sprinkler wire rather than the recommended twisted pair.

FulvioSpelta, 2019-07-29:
Just for discussion.

Maybe useful to define and use a simple and open mqtt based protocol, something like tasmota fw.

Using this approach the plugin can be generic allowing the development of any kind of sensors with very simple integration.

The mqtt broker, if not already present on the lan, can be hosted by the raspi that host SIP.

The only constraint remain the use of tcp/ip based communication but IMHO the presence of a WiFi coverage may be an assumption.

catkiller007, 2019-08-01:
Hi,

What i'm doing right now it's a pump controller for wells and tanks
Block diagram: 
[image]

The hardware and software for this it's almost ready, but what i'm trying to decide it's what interface to use for communicating with SIP. 
If i choose Ethernet then i will have to use MQTT and a switch, in IP66 enclosure for protection.

On the other hand if i go with can bus, which i have more experience with, then i will have to integrate socketCan in SIP [https://github.com/linux-can/can-utils](url). Also in this case it will be easier to wire multiple sensors/modules that are far away from SIP module

Regarding SIP interface, i will do it like this, at least for testing, displaying the values from the pump controller in the left side of the page under the clock and date.
[image]

Dan-in-CA, 2019-08-01:
Looks like you have made a good start. I like the fact that you will be using a wired buss.

The reason I like using MODBUS over RS-485 is that there is a great Python MODBUS library:
https://minimalmodbus.readthedocs.io/en/master/

It only takes a few lines of code to implement the MODBUS prorocol in Python:
`import minimalmodbus`
`valve = minimalmodbus.Instrument('/dev/ttyUSB0', 7)`
`valve.serial.timeout = 0.3`
`valve.write_register(1, duration * 1000, functioncode=6)`

There is also a good Arduino library:
https://github.com/jecrespo/simple-modbus/tree/master/Modbus%20RTU%20libraries%20for%20Arduino/SimpleModbusSlaveV10 

You can connect the Pi to the serial bus with an inexpensive USB adapter and not use any GPIO pins.
[image]
On the Arduino side the adapter costs about $1.00 US and can be connected to the TX and RX pins on the Arduino if it has 2 hardware serial ports or to other pins using a serial library.
[image]
I usually put the remote node into a weather proof electrical junction box for a cheap water tight enclosure. 
[image]

catkiller007, 2019-08-01:
Hi Dan,

The reason i was proposed can-bus is that i already have some nodes working with can-bus like this motorized valve:
[image]

Also, now im having some STM32 boards that im playing with, and on these i have a can-bus bootloader that uses UAVCAN protocol, which it's very useful when updating firmware in remote locations  
[image]

[https://uavcan.org/GUI_Tool/Overview/](url)

Maybe we can use both can-bus and MODBUS :)

Dan-in-CA, 2019-08-01:
Cool!
If CANBUS already interfaces with your hardware we should use it for sure.
There will probably need to be separate SIP plugins for each protocol but they do have some things in common.

catkiller007, 2019-08-19:
Hi,

I've just finished my pump controller based on STM32F1, it uses UAVCAN over can-bus.
The next step would be to make a plugin that reads all the uavcan.equipment.pump.Status mesages and read/configure parameters  

This is a live picture with system working.
[image]

Stifler6996, 2019-11-28:
I just wanted to throw 2 cents in to this topic. (Its been a while since I reviewed whats happening with SIP)
This idea of a modular controller to operate and read remote devices is a great idea. I immediately realized that this could expand physically 4000 feet using RS-485 (and MODBUS), no boosting required. (farms will love this!)

You have a working CANBUS which is also great. I'd like to only comment on the design of CANBUS for your future testing later. It is in the description of the BUS `"a robust vehicle bus standard designed to allow microcontrollers and devices to communicate without a host computer"`

I worked with CANBUS in the sweepers you see on the roads, CANBUS not only ran all the truck specific systems, but also the huge vacuum (that whole collector box is at reduced atmosphere!), water, pneumatic and a complex Power Transfer Unit (and you just thought those sweepers were just a brush on a motor!)

It seems to me, that CANBUS is overkill and most of the core (slave/slave communications) and to some extent fails to offer a key winning feature - distance!

Great work with this, I look forward to seeing it in operation, and I probably have a use for it myself.
👍

adrianbostan, 2021-06-03:
Hello all,

After a long pause, i would like to resume work on this project :)

closed 2021-06-13 by Dan-in-CA