Dan-in-CA/SIP #225: weather base water level isssues: no such file or directory..

aocana, 2019-09-18:
any idea why I have this error after try to configure weather adj.

> 
> Weather-base water level encountered error:
> Traceback (most recent call last):
>   File "/home/takezo/SIP/plugins/weather_level_adj.py", line 97, in run
>     today = today_info(self, options)
>   File "/home/takezo/SIP/plugins/weather_level_adj.py", line 313, in today_info
>     if name in os.listdir(path):
> OSError: [Errno 2] No such file or directory: './data/weather_level_history'

[image]

thanks for your support!

Dan-in-CA, 2019-09-18:
The current version of the plugin has a bug.
It requires a folder named weather_level_history to be in the SIP/data/folder. The plugin is being updated and a new version should be ready in a few days.
See the issue on the SIP plugins repo:
https://github.com/Dan-in-CA/sip_plugins/issues/35

aocana, 2019-09-18:
i look forward to the update!!!
thanks @Dan-in-CA  for your support!!!

aocana, 2019-09-18:
where i can see when is delivery this new release?

thanks

Dan-in-CA, 2019-09-18:
I will post an announcement on the SIP GitHub plugin forum:
Dan-in-CA/sip_plugins#35

and on the SIP forum:
https://nosack.com/sipforum/index.php

aocana, 2019-09-19:
cool!

Dan-in-CA, 2019-09-20:
OK. The weather_level_adj plugin update Is now available.
Click the "help" button on the plugin page to open the documentation page. It contains information about using geographic coordinates for your location.

closed 2019-10-26 by Dan-in-CA