Dan-in-CA/SIP #227: Stable - lcd_adj will not save changes

PiDocJohnson, 2019-11-14:
lcd_adj is not functioning, no errors or log entries

Dan-in-CA, 2019-11-14:
Thanks.
It was just updated by a contributor and I haven't had a chance to test it yet. I will dig out my old lcd and test it tomorrow.

Are you running under Python 2 or 3?

PiDocJohnson, 2019-11-14:
python 2, 3 had some syntax errors when I tried it

Dan-in-CA, 2019-11-14:
OK. The problem with saving the settings is fixed.
If you re-install the plugin it should work under Python2. 
For some reason it causes SIP to delay starting a minute or two before the plugin settings have been saved. I haven't figured that one out yet but once the settings are saved SIP starts normally.

SIP is about to have a major upgrade to work under both Python 2 and 3. The plugins are being updated to be compatible with both versions of Python.

The lcd_adj plugin needs to be updated for Python3 compatibility and this may break it for the current version of SIP. 

The version of SIP that can run under both versions of Python is currently the Py3_WIP branch. It will be merged into the main branch before the end of the year when support for Python 2 will end.

Things are a bit messy right now do to the changes.

Dan-in-CA, 2019-11-14:
Update:
I updated the lcd_adj plugin so that it runs under the current version of SIP and also under Python3.
I added smbus2 (renamed to smbus in the SIP directory). This works better and is easier for a newbie to use.

I suspect that the pylcd module is causing the delays. I would like to switch to the rplcd module but that will require a major re-write of the plugin. It will happen, just not sure how soon.

PiDocJohnson, 2019-11-14:
Well, I was better off and didn't know it (;0) uninstalled lcd_adj, rebooted, installed lcd_adj and rebooted.
System is now hung on the base screen, loading ... no response from any inputs

PiDocJohnson, 2019-11-14:
ALARM from Example_Sender!: Just an example!
Error: [Errno 121] Display not found at address 0x20
LCD plugin is active
plugins loaded:
  lcd_adj
  mobile_app
  plugin_manager
  proto
  signaling_examples
  system_update
zones changed
[0, 0, 0, 0, 0, 0, 0, 0]
 Starting timing loop

http://:::80/
::ffff:192.168.1.67:60189 - - [14/Nov/2019 14:29:20] "HTTP/1.1 GET /" - 200 OK
::ffff:192.168.1.67:60189 - - [14/Nov/2019 14:29:21] "HTTP/1.1 GET /static/images/favicon-32x32.png" - 200
::ffff:192.168.1.67:60264 - - [14/Nov/2019 14:31:12] "HTTP/1.1 GET /" - 200 OK
REFRESH yields the GET /" 200 OK line 
No CLOCK display
Your thoughts...

Dan-in-CA, 2019-11-14:
I see it.
The program is not loading fully. There is a URL for the plugin settings page that is not formatted correctly. But I don't see where the problem is coming from .  The plugin has the URL formatted correctly but looking at the home page source in the browser it is formatted differently.

Dan-in-CA, 2019-11-15:
The best solution might be to move to the newer version of SIP.
If you use the command
git checkout Py3_WIP
from the SIP folder it will load the update and the lcd_adj plugin should be running.

PiDocJohnson, 2019-11-15:
DONE! Should have done that to begin with ... Thanks Again

Dan-in-CA, 2019-11-15:
Thanks for your patience and feedback.

I think I have now found a solution that works under all versions of SIP. 
Other plugins will probably have problems for a short while but there is definitely progress.

closed 2019-11-24 by Dan-in-CA