Dan-in-CA/SIP #229: Fresh install - relay_board module hangs up webpage

Stifler6996, 2019-11-28:
Hi again!
Fresh install of Raspbian and SIP. (Using the setup from my previous issue #228)

Everything goes according to plan until I _enable_ the relay_board plugin.
The module installs correctly, but the moment I enable it (and the system restarts the service) I can no longer react with the web page. The page comes up - but I no longer enter menus etc.

Where to start - I guess it has something to do with the web page(s) that are added to the core when the module is enabled (where you used to be able to configure the relay_module items).

Dan-in-CA, 2019-11-28:
SIP and the plugins are going through a major update to make it compatible with Python3 and also Python2.
At this point the relay_board will work with the Py3_WIP branch but unfortunately not the current master branch on GitHub.

The solution is to log into the Pi, cd to the SIP directory and run the command:
`git checkout Py3_WIP`
That should allow the plugin to work.

There is also a change in the format that irrigation programs are stored in. There is an automatic conversion but that is still in the testing stage so check that your schedules are correct. The old format file is preserved in case of problems.

Dan-in-CA, 2019-11-28:
More info:
I just tested the relay_board plugin and found that the settings did not save preperly.
When I tried to set the plugin for 4 relays and high trigger level the high setting was saved but the number of relays stayed at 1. 
When I re-set the number of relays to 4 and saved the settings again the number of relays was saved.
Strange, working on it.

Dan-in-CA, 2019-11-28:
Update:
The Py3_WIP has now been merged into the main branch.
You can just run the command:
`git pull`
in the SIP directory to update your installation.

Stifler6996, 2019-11-28:
Hi Dan!
Thanks for the quick replies. I did as you said (I left the 'inoperative plugin' enabled) and just git pulled the latest, and reboot for luck.

It appears to have fixed the problems. I now have 11 channels functioning in manual.

I had, after many years, a failure of an SD card. So I decided to bring this system up to date and build it fresh (not that complicated after all). So I will be recreating all my schedules, and wont be able to check that component.

I do now have currently: a working RPi2B+ with Buster. Full IPv6 support with full domain.
SIP is happy to chat on IPv4 or IPv6. I have removed the proto, signaling example and mobile plugins (mobile didn't work anyway - or had that been fixed?)

The next thing will be to bring the mqtt module back in (I have installed and enabled with success since this current pull 👍 ), and will configure and test operation later.

As for this specific issue - I think we are done!

Great work Dan.

Dan-in-CA, 2019-11-28:
Great! 
It has been (and still is) a hectic time trying to get everything working together.
Let me know if you encounter any other issues.

I still need to update the documentation and some of the lesser used plugins.

closed 2019-11-28 by Dan-in-CA