Dan-in-CA/SIP #230: Possible time ambiguity at end of daylight savings time

karlkec, 2019-12-23:
Looking at sip.py, the time reference appears to be made with local time, converted to UTC:

`gv.nowt = time.localtime()`
`gv.now = calendar.timegm(gv.nowt)`

This could result in a UTC negative time jump of 1 hour when local daylight savings time ends. For example, using the Python 3 shell, a one-second local time advance results in a 3599 second shift back in UTC:

`>>> t = 1572771599   # Timestamp for 11/03/2019 08:59:59 UTC`
`>>> timegm(time.localtime(t)) - timegm(time.localtime(t+1))`
`3599`
`>>>`

It appears that gv.now (UTC) is used for the majority of time references. A better approach might be the reverse order:

`gv.now = time.time()`
`gv.nowt = time.localtime(gv.now)`

Then local time will jump but UTC will not.

Dan-in-CA, 2019-12-24:
gmtime() converts a timestamp to a  struct_time in UTC in which the dst flag is always zero.

However calendar.timegm() converts a  struct_time (returned from time.localtime()) to a timestamp.
It is a way of getting local time as a timestamp in Python.

karlkec, 2019-12-24:
OK. Sorry for my misunderstanding. I thought gv.now was intended to be UTC.

Dan-in-CA, 2019-12-24:
No problem.
calendar.timegm() is a fairly obscure function and is easily confused with time.gmtime()
They are the inverse of each other.

closed 2019-12-28 by Dan-in-CA