Dan-in-CA/SIP #231: Station are energized upon starting sip.py

elongo, 2020-03-11:
Hi. I wonder if anyone here knows why, when running sip.py the stations (valves) are immediately energized all at the same time and with out a station program indicating so. 

Right now, when I run sip.py the valves are turned on immediately, and not only doing the normal solenoid click, but keeping it high (making permanent sound).

Any clues how can I invert this?

Thanks.

Dan-in-CA, 2020-03-11:
I would need more information about your configuration.
Such as are you using a relay board? 
Be sure the Options > station handling > active low relay box is not checked. It inverts the station logic and is for use with relays connected through shift registers.

kenbloch, 2020-03-20:
I am assuming you installed from the SIP application the relay board plugin. After the plugin is installed you must click on the "relay board" plugin under the plugin menu and then configure for how many relays.

MartinMMx, 2020-11-05:
> 
> 
> [...] make sure you are not using the lite version of the Raspian software as some features are missing that have an impact on using relays.

Can you elaborate what features are missing there? I hope they can be added with apt?
And it would be great to have this information in the documentation - I already installed rasbian lite now. :(

Dan-in-CA, 2020-11-05:
The only missing features I can think of are git and the optional pigpio library.
It you are running SIP under Python3 you will also need to install the Python 3 version of RPI.GPIO if you are using that (default) library.

I runs SIP on the lite version of Raspberry Pi OS with out a problem.

MartinMMx, 2020-11-05:
OK, thanks for clarification. This explains why it also works here: even RaspberryOS 10 still uses Python2.7 by default.