Dan-in-CA/SIP #234: Flow meter plugin revisited for SIP4

mattip, 2020-04-19:
xref gh-143 and @DaveSprague

I recently updated my SIP system to the raspberryPi4, with the Seeed studio relay hat and the cli_command plugin using [this cli package](https://github.com/johnwargo/Seeed-Studio-Relay-Board). Since I want some feedback whether the system actually worked, I added a hall-effect [flow meter](https://www.seeedstudio.com/G3-4-Water-Flow-Sensor-p-1083.html). Note I do not need accurate-to-the-last-drop readings, rather confirmation that the programs ran. I used the plugin from gh-143 and @DaveSprague (modified for SIP4) to
- add the plugin and a basic setup page for the flow sensor plugin
- use a `multiprocessing.Process` to set a GPIO event loop that increments a shared counter on each pulse from the flow meter
- starts a thread to update flow sensor readings once every 3 seconds from the number of pulstes on the shared counter. The update updates data in `gv.pluting_data['fs']` (`fs` for flow sensor)
- Use the SIP signal module to set a `stations_scheduled` callback to update the program html page with current program flow data, under the assumption that only one program is active at any given time
- Hack the log json code to add a column for the flow data

The last point (hacking the log code) was the only one that requires some adaptation from the basic SIP code. Maybe SIP could provide a nicer API for modifying the program html page, maybe it is fine the way it is now. 

After all that, my questions are:

- @DaveSprague would you like help updating the flow sensor plugin to SIP4? Even without the logging hooks it can show data on the program page while they are running, to give some confirmation that everything is working as designed.
- Would it be possible to provide logging hooks to allow plugins to add columns to the logfile?

Dan-in-CA, 2020-04-20:
Hi Mati,
I have been working on an update to the core SIP code that allows plugins to display data such as sensor data on the home page. Either in the timeline next to the station name or in a bar in the page footer.

[image]

There is also an option for adding data to the run log.

[image]

It is still a little buggy but use-able. I will work on adding a development branch to this GitHub repository in the next couple of days so you can test it out.

Dan

mattip, 2020-06-22:
> I will work on adding a development branch to this GitHub repository in the next couple of days so you can test it out

@Dan-in-CA did it happen?

Dan-in-CA, 2020-06-22:
Not yet sorry,
Had to take care of some other issues.
Thanks for the reminder.

Dan

Dan-in-CA, 2020-06-22:
OK. 
I pushed a new branch to the SIP GitHub repo. it is named **plugin_data**
Be sure the **proto plugin** is enabled and you can run a program or use Run Once to see example data on the timeline.

This should add a column to the log on the Log page

There should also be a new field in the footer of the UI pages with some example data.

Starting at line 45 of proto,.py in the plugins folder are code examples that control the data displays.

Dan

Dan-in-CA, 2022-08-18:
There is a new flow meter plugin called Flow. It is being updated to include notifications of problems.

closed 2022-08-18 by Dan-in-CA