Dan-in-CA/SIP #235: Mobile App iOs issue SIP v4.0.27

Bob2345de, 2020-04-24:
Hi,

Tried updating a very old version 3.0.13 running on Jessie and ran into an issue with the mobile app not connecting. Have now tried a fresh install on the latest version of Buster but run into the same error.

The app’s device scan finds the new system, but does not want to connect. The console shows the following error (same as it was on the old OS) the app (newest ios version) still connects fine to my old install. would appreciate any help!

http://192.168.3.116:8080/
192.168.3.133:54336 - - [24/Apr/2020 20:59:28] "HTTP/1.1 GET /jp" - 200 OK
192.168.3.133:54336 - - [24/Apr/2020 20:59:28] "HTTP/1.1 GET /jn" - 200 OK
192.168.3.133:54336 - - [24/Apr/2020 20:59:28] "HTTP/1.1 GET /jo" - 200 OK
192.168.3.133:54336 - - [24/Apr/2020 20:59:29] "HTTP/1.1 GET /js" - 200 OK
Traceback (most recent call last):
  File "/home/pi/SIP/web/application.py", line 289, in process
    return self.handle()
  File "/home/pi/SIP/web/application.py", line 280, in handle
    return self._delegate(fn, self.fvars, args)
  File "/home/pi/SIP/web/application.py", line 530, in _delegate
    return handle_class(cls)
  File "/home/pi/SIP/web/application.py", line 508, in handle_class
    return tocall(*args)
  File "/home/pi/SIP/plugins/mobile_app.py", line 84, in GET
    u"ct": get_cpu_temp(gv.sd[u"tu"]),
TypeError: get_cpu_temp() takes no arguments (1 given)

Bob2345de, 2020-04-24:
You can close this issue as I have just read the other topic which was opened and closed in period I was working on this.

One note: would it make sense to remove the mobile app plugin from version 4 as it doesn’t work and might lead others to believe that it does...

closed 2020-04-24 by Bob2345de

Dan-in-CA, 2020-04-24:
The reason for keeping the mobile app is that some users access the output for other uses.