Dan-in-CA/SIP #236: Scheduled programs are not starting.

elongo, 2020-05-10:
Hi. 
I wonder if some one can give me a hand on this one.

I have been working on a new branch of the sip library, which is under test for now, and I am experiencing the following issue:

The scheduled programs are not being executed.

The program is created, and it appears in the interface as about to be executed, but then the stations are not turned on and the timer count down in the UI doesn't start either. Is frozen.

I do "RUN NOW", the same issue is experienced.

In my script  sip.py, the lines where programs are suppose to be execute, turning on stations, looks as follows (NOTE: the functions, device_on(sid) and device_off(sid), were created by my self and have proofed to work properly, in the RUN ONCE option):

`if gv.sd['bsy']:
            for b in range(gv.sd['nbrd']):  # Check each station once a second
                for s in range(8):
                    sid = b * 8 + s  # station index
                    if gv.srvals[sid]:  # if this station is on
                        if gv.now >= gv.rs[sid][1]:  # check if time is up
                            gv.srvals[sid] = 0
                            set_output()
                            gv.sbits[b] &= ~(1 << s)
                            device_off(sid)
                            if gv.sd['mas'] - 1 != sid:  # if not master, fill out log
                                gv.ps[sid] = [0, 0]
                                gv.lrun[0] = sid
                                gv.lrun[1] = gv.rs[sid][3]
                                gv.lrun[2] = int(gv.now - gv.rs[sid][0])
                                gv.lrun[3] = gv.now
                                log_run()
                                report_station_completed(sid + 1)
                                gv.pon = None  # Program has ended
                            gv.rs[sid] = [0, 0, 0, 0]
                    else:  # if this station is not yet on
                        if gv.rs[sid][0] <= gv.now < gv.rs[sid][1]:
                            if gv.sd['mas'] - 1 != sid:  # if not master
                                gv.srvals[sid] = 1  # station is turned on
                                set_output()
                                device_on(sid)
                                gv.sbits[b] |= 1 << s  # Set display to on
                                gv.ps[sid][0] = gv.rs[sid][3]
                                gv.ps[sid][1] = gv.rs[sid][2]
                                if gv.sd['mas'] and gv.sd['mo'][b] & 1 << (s - (s / 8) * 80):  # Master settings
                                    masid = gv.sd['mas'] - 1  # master index
                                    gv.rs[masid][0] = gv.rs[sid][0] + gv.sd['mton']
                                    gv.rs[masid][1] = gv.rs[sid][1] + gv.sd['mtoff']
                                    gv.rs[masid][3] = gv.rs[sid][3]
                            elif gv.sd['mas'] == sid + 1:
                                gv.sbits[b] |= 1 << sid
                                gv.srvals[masid] = 1
                                set_output()
                                device_on(sid)

            for s in range(gv.sd['nst']):
                if gv.rs[s][1]:  # if any station is scheduled
                    program_running = True
                    gv.pon = gv.rs[s][3]  # Store number of running program
                    set_output()
                    device_on(sid)
                    break
                program_running = False
                set_output()
                device_off(sid)
                gv.pon = None`

Dan-in-CA, 2020-05-10:
I see a possible problem.
The function "set_output()" is what actually turns stations on or off.
It should probably be after your device_on(sid) or device_off(sid) function calls.

such as:
```
device_on(sid)
set_output()
```

elongo, 2020-05-19:
Thanks for the feedback... seems that there was an typo in session.py.

But now what I experience, is that the programs created are not being scheduled in the calendar, and therefore executed in the determined time (at any time, actually). Wondering where could be the cause for that!!

Thank you!

e

Dan-in-CA, 2022-08-18:
Closing 2 year old issue.