Dan-in-CA/SIP #238: Connection to Raspberry

NJMoz, 2020-05-22:
Hi Dan trying to contact you The web forum registration page does not accept my email neil@uk-esc.com I have tried your software but I don't seem to get the relays to switch on 
I wired it up as per the shift register drawing but nothing switched on . The web interface works fine and the relays I programmed in switch correctly and the log works but on the Raspberry Pi 3 nothing doing. 
Do I need to set the different variables, 
do you have a better explanation of the software and connection diagrams 
does one have to use the shift register as I only have 5 relays to switch on and off every 20 min in a sequence  
I want to use the timer on my aeroponic systems and don't mind contributing if I can get it going

Dan-in-CA, 2020-05-22:
Hi Neil,

If you are using only 5 relays you can use either the relay_board or relay_16 plugin and connect the relays directly to the Pi.
In SIP go to Plugins > manage plugins > browse more plugins and install the plugin you wish to use.

For the relay_board plugin see the documentation at:
https://github.com/KanyonKris/relay_board/wiki

For the relay_16 plugin, after installing the plugin open the plugin's  setup page from the Plugins menu and click the HELP button to open the documentation page which contains a wiring diagram.

Relays can have either high level trigger or low level trigger and that needs to be set in SIP for the relays to operate properly. That may be part of the problem.

I'm not sure why the forum is not accepting your email but it is fine to post questions here.

You can set SIP to switch the relays on and off every 20 minutes in the program set up. Select recurring at:
Programs > add a new program > single pass / recurring > Every. Set the time for 20 min.

Let me know how it goes.

Dan

NJMoz, 2020-05-22:
Hi Thanks for replying  I got it up and running

 found the relay board solution in your issue archives

Kind Regards

Neil Jackson (Mba)
+44 (0)20 3239 9074
+27 (0) 71 5090 107
Skype njchongoene
neil@uk-esc.com
www.uk-esc.com

*We Develop Your Business & Your Future*

closed 2022-08-18 by Dan-in-CA