Dan-in-CA/SIP #243: How many zones

otterdog, 2020-07-10:
Sorry, but I can't find this information anywhere... and I don't know where else to ask. 

Ultimately, how many zones can this have?   I'm in the research stage of building an aeroponics farm... and will have about 400 relay (zones)  is it possible to use this software?  Thank you Dan.  - Rob

Dan-in-CA, 2020-07-10:
Hi Rob,
Go to Options > Station Handling > Station extensions
and increase the number of extensions. Each increase of one adds 8 stations. There is no limit to the number of stations but I have not tried 400. It may slow things down a bit.

You could also have multiple controllers that each handles, say 100 relays each.

Good luck with your project.

Dan

otterdog, 2020-07-10:
Thanks.. that help.  As I haven't actually installed the program.. and just doing research right now to see what's my best choice, your info helps. 
Just one more question if you don't mind.   When you say 'multiple controllers' are you talking about multiple Raspberry Pi's connected to one another .. or.. ?
Hope all is well out there in CA... i'm from San Diego myself.  Glad I live in Vietnam now with all that's going on Virus-wise out your way :(   

Thanks Dan.

Dan-in-CA, 2020-07-10:
Multiple controllers could be Raspberry Pis connected together or running independently. SIP is pretty flexible in how things can be configured.
There is also the possibility of developing a plugin that can handle unusual situations.

Yeah the virus is raging here. We are also expecting extreme heat this weekend. the National Weather Service predicts a high of 126F in Death Valley

otterdog, 2020-07-10:
Ouch! I lived in Temecula.. so it got hot there also.. but not THAT hot :)    Cheers, Dan.  Thank you for your info.. it's greatly appreciated.   Stay safe out there.  Hopefully with your help ( and other people's as well ) we'll build out this Aeroponic farm in December.

closed 2022-08-18 by Dan-in-CA