Dan-in-CA/SIP #250: Incompatibility with python3.9

Tails86, 2020-11-02:
https://github.com/Dan-in-CA/SIP/blob/8a1f439f3cdc61a7a3d93fca9775fca7e6e4a605/i18n.py#L66

Changing this line to the following allows SIP to be run in Python3.9. 
```
    install_kwargs = {}
    if sys.version_info.major == 2:
        install_kwargs['unicode'] = True
    gettext.translation(u"sip_messages", localedir, languages=[ui_lang]).install(**install_kwargs)
```

Otherwise, I get this error:
Traceback (most recent call last):
  File "/home/pi/SIP/sip.py", line 12, in <module>
    import i18n
  File "/home/pi/SIP/i18n.py", line 66, in <module>
    gettext.translation(u"sip_messages", localedir, languages=[ui_lang]).install(True)
  File "/usr/local/opt/python-3.9.0/lib/python3.9/gettext.py", line 354, in install
    for name in allowed & set(names):
TypeError: 'bool' object is not iterable

Dan-in-CA, 2020-11-02:
Thanks.
SIP has been updated.

Tails86, 2020-11-03:
That works! Thanks

closed 2020-11-03 by Tails86