Dan-in-CA/SIP #251: relay_16 does not appear to work with SIP4

Stifler6996, 2020-12-01:
Hi,
I upgraded my install from a version 3 revision to SIP4, and none of the sprinkler zones would work after.

So since it was a good opportunity to switch to Ubuntu 20, I build a fresh SD card, and cloned the latest SIP4.
I added the relay_16 module and "restarted sip"

Several things I noted. restart, doesn't work. so I ran "sudo service sip3 restart"
Next I noticed that relay_16 does not come up in the Plugins menu (when you click the button is lists your plugins if they have options, like the MQTT ones). I know there wasn't anything to configure with relay_16, but I believe it was still listed because it had a helpful page that gave pin connections.

My my build is Ubuntu 20.04-1
Running Python 3.8.5

Not sure where to start looking.

Dan-in-CA, 2020-12-01:
You need to install the Python3 version of the rpi gpio library.
sudo apt install python3-rpi.gpio

Stifler6996, 2020-12-01:
Yes, just identified that. I didn't realise that py2 and py3 were different.
And I totally forgot to do that step in the new build.

ty. I will make a new bug report. I am seeing other problems unrelated.

closed 2020-12-01 by Stifler6996