Dan-in-CA/SIP #258: Remove Google Font

scharc, 2021-05-19:
I am running SIP in my allotment (off garden) with an LTE Router as internet gateway. Sometimes the connection is very bad depending on the weather so I would love to see SIP without any external dependencies because it will block page loading. 

If needed I can provide a pull request for the coresponding css file.

Dan-in-CA, 2021-05-20:
I agree that eliminating external dependencies is a good thing.

Just wondering:
Is the browser not caching the fonts?
Have you been using your modified css for a while?

Go ahead and make the pull request. I'll take a look at your solution.

Dan

Dan-in-CA, 2021-05-20:
I did some tinkering with SIP's base.css file. 
Replacing
`@import url(https://fonts.googleapis.com/css?family=Quicksand:300,400,700);`
with
`@font-face {
  font-family: Quicksand;
  src: url(./fonts/Quicksand-VariableFont_wght.ttf);
}`
and adding a local copy of the .ttf font. That seems to be working.
would that work for you?

Dan

scharc, 2021-05-20:
Great! That would have been my PR to fix the issue.

Dan-in-CA, 2021-05-20:
The update has been pushed to the SIP repo.
I actually added several font files to handle different weights. The variable weight file didn't seem to work properly

Dan

closed 2022-08-18 by Dan-in-CA