Dan-in-CA/SIP #261: Display water pressure in home page

adrianbostan, 2021-06-07:
I've been playing around with the with the pump control plugin, and i believe its useful to display the water pressure and some other parameters on the home page. Any hints on how to do that ?

Dan-in-CA, 2021-06-07:
Take a look at the "plugin data" branch of this repo. It has a couple of ways to display that sort of data.

adrianbostan, 2021-06-07:
Tried to use the proto plugin from plugin_data branch (replaced webpages.py also) but i miss something because its not working.

`
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/SIP/plugins/proto.py", line 50, in footer_test
    example = showInFooter() #  instantiate class to enable data in footer
  File "/home/pi/SIP/webpages.py", line 768, in __init__
    self._idx = len(gv.pluginFtr)
AttributeError: 'module' object has no attribute 'pluginFtr'
`

Dan-in-CA, 2021-06-08:
Hi Adrian,
I cloned a fresh copy of SIP, changed to the plugin_data branch from the SIP directory with:
`git checkout plugin_data`
Opened the plugin manager and enabled the proto plugin and the data display appeared below the timeline on the home screen. I tried it with both Python 2 and 3.

There is another data display that can appear on the timeline next to the remaining time for the running station. This requires an irrigation program to be set up and run. You can use the "run now" button on the programs page for testing.

Dan

adrianbostan, 2021-06-08:
Hi Dan,

I've tried this way and it works, the only issue i found its the version:  `Software version: 4.0.27 (2020-04-20)` and the update plugin doesn't work. Basically it would be great if we can use the proto plugin from Plugin_data with the latest rev of SIP.

Adrian.

Dan-in-CA, 2021-06-08:
Yes, There have been many updates to the master branch after the plugin was posted.
I will work on updating the plugin_data branch. 
It may take a few days and I will post a message here as soon as it is ready.

Dan

adrianbostan, 2021-06-08:
Hi Dan,

So basically what I'm trying to do, is to make a custom pump_controller plugin and display some data on the home page. 
For that i modified the original pump_control.py to display the data in the footer.
The only issue i see with this one is that, even if the Pump Status changes, i have to refresh the page to updated the value, the upper one (Pressure) works as it should.
![enter image description here](https://lh3.googleusercontent.com/NIfyi0qIWlDX434j1686Csir4oZ1-mBv-lQO4PC9Fne-9E4RJFuZ0eBTmCXi4AaG0PSxlsVaMQj8iFX2_srQLop8uD3Cgg_hI1UF7uO7R0tWYpaVONIlH-sI1XQY_W37WFp1FFoEvAYnUrxx7p_2Wdeohnl0hSDj5-hDFVRYYGWXtoYJC-gwvlyLiDldngLuPAW8_r1tNfR6PW7Uf_JTkwnefMF2GRLsf91UcDB4FGGqnE8NJM0r4oAxfoL4hMXftond50QzZPQCnr7bpNHVPoRbWMfJ-1phnnmOi-fkYCjb6tu6Hq-ppBxua0dGQVdMni5mhQmFTQ1zbDuPy9JYS8DBsbeysqAEYcrehhoMiVZDuL9yb3ZBhk6X7EVV34iAh9S8k7JK2L53QCusyGufeJxxYNKBS9wmKUjkwE1nXT6iGY1R2SluF-RCFkmXMo6IHqBJYUtaVqPp2lH4fXPbLKs8Jo_SpCHGNtM-5gVQUS643ZgcfIAFPf9zlItL00MF7T20WffpsPfS3vQy0JwarvMufkmVpvDbAS5iE_WD8dN1mO6CQWRyiBa-8OINyoeHj3zchZ4gq2zDq0gnSyQubwUJVR7fJD32x1ByOFljzCIMI2Atk9ajBQfccqVza5fZvgPSadSjl-OlkI2HPq4EulUICrSrUrSnz1av5VdCbYahMEAgE9l0nAhZ2SGD2iR3qTSQJ_t7X-rfMIkXFkq2fopI=w1844-h827-no?authuser=0)
This is the mobile version:
![enter image description here](https://lh3.googleusercontent.com/Z5jSjFeI57t7IqQDhXyLo6eA2RKx23mXU2kivdddfyOV2JRmo8gjA6SYWRT3sQPDPSG9VrzGD3ug9NgDKjuJPfdKwwg_Zv2V6nfFV3zQy2WuCE1aKrMPBBkJgGvdI_orMAFlI5tqB-bP758oWeK19TLPQZRNx4sqlT8kkmJjTi2Id0h-5ABJtWnb06BNneXbmnvqIZSbLLcbJwbrStZ5Ntb83oAMeNCFUs5iMf4fGiY4SrPWUbd_kh-a7Xm1Vh6od-KxwNH3ZAY1hV_USUb0YyG62lsiQPdZLUCf5vVjb3UwdS1SxYCZLVIZnVy5PH5K3ehoGEwQwBFd1Yysd3xHJ9Qgy2HzY_2EmXUxcLY5P1TZH1PwdqsIqV1SHRYHZBxFeaKK68uyx_3mE9P-Kj8565Mm6TiaKuRNyPJTxgfK_F8uThCh1eGv1hHRRHDB-0bGGRucGjPvyj69KQyKaxcFRG3WiPzbdA0bgeRsvhbBuhOGqLXrcafq-zcdBF7ak5_DbXWabHBwqBQaN5N1GIBpqxkjvggOYxOGhmX3dXKKu--2RcubXop468f8l5k9fQggx2MHCI6ihkD95xWWnG9RPWtSQjblfylvOaKmlKo5ukjOPxjRBXVAAKla10aCsNzzU27RcbTzvVnN25z1AJgUSaWEdABocfp7SpDmNjnLLUT7A-nStAw6pqnCaJlS7c5TPV1IPiI2dhXF8nEH80EGuyIp=w738-h1600-no?authuser=0)

    def  footer_test():
    
    """
    
    Example of plugin data display in UI footer.
    
    Run in background thread.
    
    """
    
    water_pressure = showInFooter() # instantiate class to enable data in footer
    
    water_pressure.label =  u"Pressure"
    
    water_pressure.val =  0
    
    water_pressure.unit =  u" mBar"
    
      
    
    pump_status = showInFooter() # instantiate class to enable data in footer
    
    pump_status.label =  u"Pump Status"
    
    pump_status.val =  0
    
    pump_status.unit =  u" "
    
      
    
    while  True: # Simulate plugin data
    
    water_pressure.val = get_now_pressure() # update plugin data
    
    pump_status.val = get_now_status() # update plugin data
    
    print('Pressure',water_pressure)
    
    print('Status',pump_status)
    
    sleep(2)

Dan-in-CA, 2021-06-08:
I am not getting the images.

It looks like both values should update. Is the get_now_status() function returning updated values?
Do the print functions show updated values?

What happens if you try to show the status only? does it update then?

adrianbostan, 2021-06-09:
Solved images issue.
The get_now_status() its working properly because if i show only the status in the footer, it works. 
Also the print functions are displaying updated values.

Dan-in-CA, 2021-06-10:
I am getting close to a fix.
the code the updates display with the plugin data is in base.html starting at about line 157:

             function dataTimer() {
                // Call to get plugin data
                jQuery.getJSON("/api/plugins", function(piVals) {
                	var parID = 'piData'  + piVals.fdata[0][0] 
                	jQuery('p#'+parID ).find('span[name="val"]').text(piVals.fdata[0][1])
                } 

It only updates one paragraph in the display.  I will be changing it to update all paragraphs. I am also making some other changes to the related code.

It may be a day or two until it is ready.

Dan-in-CA, 2021-06-11:
Adrian,
I made a new branch named "PID_footer"on the SIP repo  (PID = Plug In Data) .

It is a branch of the current SIP rev with the plugin data displayed in the footer. there are 2 example data sets in the Proto plugin that both update without needing a screen refresh. It should work with any number of data sets.

I also tweaked the CSS to make it responsive  so it works better on mobile devices.
Let me know what you think.
Dan

adrianbostan, 2021-06-11:
Hi Dan,

Everything its working now and the mobile page its looking much better.

[image]

adrianbostan, 2021-06-11:
To explain more on what I'm trying to achieve:
[image]
 
Basically I'm writing a new plugin from the original one, that will have can-bus communication:
Example, very easy to do it in python with soketcan.
`        if water_pressure.val <= 50:`
`            msg = can.Message(arbitration_id=0x00000066, data=[ 0x22, 0x17, 0x00, 0, 0, 0, 0, 0x39], is_extended_id=True)`
`            bus.send(msg)`
`        else:`
`            msg = can.Message(arbitration_id=0x00000066, data=[ 0x22, 0x17, 0x64, 0, 0, 0, 0, 0x9D], is_extended_id=True)`
`            bus.send(msg)`
Regarding the module, I'll be using Arduino with MCP2515 for simplicity and low cost.
The plugin web page will need some changes because i will be adding some new features, like a flow sensor and a wait time after reaching a low water level.

I will post the project in a few days--weeks :)

adrianbostan, 2021-06-13:
> Adrian,
> I made a new branch named "PID_footer"on the SIP repo (PID = Plug In Data) .
>
> It is a branch of the current SIP rev with the plugin data displayed in the footer. there are 2 example data [...]

Hi Dan,

I was thinking, maybe you can add the showOnTimeline functionality to "PID_footer"on the SIP repo (PID = Plug In Data) .
Because, i can add flow sensor data to each station from this plugin https://github.com/DaveSprague/sip_plugins/tree/master/flow_sensors.

Dan-in-CA, 2021-06-13:
Yes I was already planning to add that. Just wanted to be sure the footer part was working properly.
After adding the timeline display functionality I plan to merge the PID_footer branch into the master branch.

There is also a part of the plugin data functionality that can add plugin data to SIP's run log. that needs a bit more testing.

Dan-in-CA, 2021-06-13:
OK. I am moving this issue to Discussions in the Ideas section.

closed 2021-06-13 by Dan-in-CA