Dan-in-CA/SIP #263: Feedback for damaged relay module or valve

adrianbostan, 2021-06-13:
Last week i had an issue with my system. Because the raspberry pi and the relay module were put in an area with high humidity, water got inside the box. It was a box rated with IP65, but some how water accumulated in the lower part of the box, where the 8ch relay module was. (maybe failed gasket). 
At that point i was trying to understand what happened, why the valves are not working, from the SIP the command was sent but no action on the valve. Then i opened to box, later to discover the relay board was destroyed.
Now, I'm thinking that develop a module that can provide feedback to SIP and signal if any valve are failing.

[MC33879 ](https://www.nxp.com/products/power-management/smart-switches-and-drivers/configurable-switches/configurable-octal-serial-switch:MC33879) will be a good candidate because it can detect an open circuit and over-current condition.
At this point i can design the hardware and manufacturing it and also do the firmware for it.
Probably we can make it work with multiple communications protocols like: can-bus, mod-bus, Ethernet TCP, etc.

But i will need some volunteers to help me with the SIP plugins :)

mattip, 2021-06-13:
Since the proof of work is water being delivered to the plants, I think a better strategy would be to add a flow meter to your system, along the lines of issue #234 which I never finished updating

Dan-in-CA, 2021-06-13:
Hi Adrian,
This sounds very interesting.
I sent you an email at the gmail address listed in you GitHub profile regarding converting issues to Discussions.
I would like to move issues like this one to Discussions (see the nav bar at the top of this page). It will allow for more interaction and idea sharing from the software community.
When an issue is converted to a discussion it will be closed but a note will be added to indicate that it has been converted to a Discussion.

Let me know it this is OK with you.

Dan

closed 2021-06-13 by Dan-in-CA