Dan-in-CA/SIP #269: Change SIP to other name

dioh, 2021-06-18:
I find that SIP is confusing as there is an associated technology called SIP (Sub irrigation planters).

Dan-in-CA, 2021-06-18:
But Sub-Irrigated Planter is not software.

You can change the name of your SIP software installation on the Options page: 
Options > System > System name
The new name will be displayed in the header of each page.

Noschvie, 2021-06-22:
SIP / Software : Session Initiation Protocol

Dan-in-CA, 2021-06-22:
Yes. I am aware of that also.
The problem is that changing the name would effect every part of the project. Including this GitHub repository, plugins the discussion forum and an extensive rewrite of the software code itself.

It is not going to happen.

closed 2021-06-22 by Dan-in-CA

dioh, 2021-06-22:
You are right, and I think the reason not to fix it is good. Sorry for bringing this up, I think this is a nice project!