Dan-in-CA/SIP #273: Station Grouping

garudaonekh, 2021-08-03:
Hi,
1. Can we group station together for easy naming and scheduling?
2. Can we set a group of station to be run sequentially but not all sequentially? Here's my problem: we have several greenhouses, each greenhouse has around 8 to 10 stations and we need to irrigate them sequentially.

Thanks;

Dan-in-CA, 2021-08-03:
Sip uses plugins to add capabilities.
There is a plugin named "combine_stations" that allows you to group stations for scheduling. You can set individual run duration for the stations in the group
You can see a list of the plugins at:
https://github.com/Dan-in-CA/sip_plugins
See the README for the list with a brief description of each.

You can view the SIP documentation at: 
https://github.com/Dan-in-CA/SIP/wiki

garudaonekh, 2021-08-05:
Thanks. That solved my problem

closed 2021-08-05 by garudaonekh