Dan-in-CA/SIP #281: Watering time, main page stuck on Loading time

AlanTHumphreys, 2021-12-26:
Hi all, having issue with watering time on main page stuck on Loading... after a fresh install the screen looks ok, but as soon as i flip
enable as service (SIP 2 or SIP3) the main screen changes to Loading... The only way i can get it back is to do a full install again.

any ideas on why it could be happening.

Regards, Ala
[image]
n

Dan-in-CA, 2021-12-27:
I am curious why the system on/off button is off. What happens of you change it to on?
Do you have any plugins installed? If so could you try disabling them.
If you start SIP from the command line do you see any error messages?

AlanTHumphreys, 2021-12-28:
Hi Dan, since the issue non of the buttons worked except for running the programs. I was still running python2, have flipped
to python3 all is great now except for email module not loading even though its enabled under plugins.

No errors from command line, Only modules loaded are the default with Email and relay_board.

sudo python sip.py i can see the email module loaded ok. but running as service its a no show in the module list , currently running Debian latest version. first issue i have had in 5/6 years of use been flawless from get go.

[image]
[image]

Dan-in-CA, 2021-12-28:
Thanks for the feedback. 
I'll take a look at the email plugin and get back to you.

Dan-in-CA, 2021-12-28:
It looks like a lot has changed since the email_adj plugin was created. The Python3 email module will require change that are not easy to make both python 2 and 3 compatible. 

What I am planing to do is leave the email_adj plugin as is for use by anyone still running under Python2 and make an updated version for Python3 only. This may take a while to develop and test but it will be more secure.
Some of the newer plugins are Python3 only and I will be working on a Python3 only version of SIP.

Dan

AlanTHumphreys, 2021-12-29:
Hey Dan, thanks for the update. shall wait for the mod.

Alan

Dan-in-CA, 2021-12-29:
Alan,
I am considering a couple of ways to increase the number of SIP events that can be selected for email notifications.
What would you like to be able to monitor via email from SIP?

Dan

noisymime, 2021-12-29:
FWIW I also had the same `LOADING...` issue after upgrading. Like Alan I was still on python 2, switching to python 3 resolved the issue.

Dan-in-CA, 2021-12-30:
Thanks, It is fixed now for anyone wanting  run under Python2.
It had to do with the new named program feature and caused a hiccup in converting the name to JavaScript format.

Dan

AlanTHumphreys, 2021-12-30:
Hi Dan, 
the main event for me was, notification the program ran. being in an area where they are always cleaning the recycled water
pipes, email notification on pressure drop would be good, or SIP would pause till specified percentage is reached.

Alan

Dan-in-CA, 2022-01-19:
@AlanTHumphreys ,
The new Python3 email plugin named "sip_email" has now been posted to the SIP plugins repo.
Let me know if it does what you need or if there are any other notifications you would like to see.  
Dan

AlanTHumphreys, 2022-01-23:
Excellent Dan, thank you will try it out

Alan

closed 2022-08-18 by Dan-in-CA