Dan-in-CA/SIP #285: How to get in touch regarding a security concern

JamieSlome, 2022-02-20:
Hey there!

I belong to an open source security research community, and a member (@noobexploiterhuntrdev) has found an issue, but doesn’t know the best way to disclose it.

If not a hassle, might you kindly add a `SECURITY.md` file with an email, or another contact method? GitHub [recommends](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) this best practice to ensure security issues are responsibly disclosed, and it would serve as a simple instruction for security researchers in the future.

Thank you for your consideration, and I look forward to hearing from you!

(cc @huntr-helper)

Dan-in-CA, 2022-02-20:
Jamie,
I sent you an email with contact information.

JamieSlome, 2022-02-21:
@Dan-in-CA - thanks 👍 

Will respond to you via e-mail now.

closed 2022-05-06 by Dan-in-CA