Dan-in-CA/SIP #287: Master Valve Indicator

gdgeist, 2022-06-21:
Main page doesnt indicate Master Valve is on for Programs or Run Once - only for manual.  Ill check and see if i can tell where it misses the update.

Dan-in-CA, 2022-06-21:
I just checked with SIP V4.1.42 and the master appears on in both run once and a program.
Do you have the stations set to activate master on the stations page?

gdgeist, 2022-06-22:
I do - screenshot of a runonce
!
[image]
[Screenshot 2022-06-22 140819](https://user-images.githubusercontent.com/13734613/175117008-162b6590-71fe-4cee-984e-fc53939e53e3.png)

gdgeist, 2022-06-22:
[image]

Dan-in-CA, 2022-06-22:
OK I see it too.
I will work on fixing the problem. It may take a couple of days. 
In the mean time can you re-wire your stations to use a station 8 or less ad the master?

Dan-in-CA, 2022-07-15:
Finally found it. That was a tough bug to track down but a fairly simple fix.
SIP version 4.1.46 has the fix.

gdgeist, 2022-07-18:
Updating - FWIW - I updated the LCD plugin for myself to show the station name instead of the station number.  Helps my wife a bit.  I didnt bother with an error handling routine if the name is empty though.  That plugin also doesnt use the C->F conversion function when displaying the CPU temp.  Forgot to fix that too this weekend but will and maybe will submit a pull request on that shortly.  Then my next task is allowing for manual runs when the rain sensor is active.  I never noticed that the off the shelf rain sensors depend on a wick to expand with moisture and then to stop saying its raining for the wick to contract as it dries.  In a humid environment that may be a day and occassionally i need to water manually i.e. do a run once/manual station even though the rain sensor is on.

Dan-in-CA, 2022-07-18:
I'm not sure if this would help but you could also set some stations to ignore the rain sensor from the Stations page.
A pull request for your LCD plugin updates would be great!

Dan