Dan-in-CA/SIP #288: HTTP GET commands - blocked by CORS policy

SlimyFrog123, 2022-06-22:
I am trying to make a simplified web interface for SIP in JavaScript, but I am being restricted by the CORS policy.

I suggest adding an Access-Control-Allow-Origin: * header to the HTTP GET API.

Dan-in-CA, 2022-06-22:
Take a look at the mobile app plugin (mobile_app.py in the SIP/plugins folder). It is enabled by default and provides access to SIP's setting using json.
It has the Access-Control-Allow-Origin: * already included (lines 95. 107, 130 etc.). It may provide what you need.

Dan

SlimyFrog123, 2022-06-23:
> Take a look at the mobile app plugin (mobile_app.py in the SIP/plugins folder). It is enabled by default and provides access to SIP's setting using json. It has the Access-Control-Allow-Origin: * already included (lines 95. 107, 130 etc.). It may provide what you need.
> 
> Dan

Here's what I'm trying to do in the web interface that I am building:
 - Show station status (on, off)
 - Allow the end user to start the stations for a specific amount of time
 - Allow the end user to stop the stations

I have some questions about this mobile app plugin:
1. How do I use it?
2. Will it do the things I want it to do?
3. What is the address that I use (eg. http://x.x.x.x/bla) or is it the same as the HTTP GET commands?
4. Is there any documentation about how to use it, or any examples that I can go off of?

Thanks in advance!

Dan-in-CA, 2022-06-23:
OK, I should have pointed you to the GET command reference:
https://raw.githubusercontent.com/wiki/Dan-in-CA/SIP/SIP_GET_commands.pdf

For example to show station status you would use # 2. Get station bits
You can check all stations or just one specific station. For example 
http://x.x.x.x/sn?sid=1 returns the status of station one (1 = on, 0 = off)

To start or stop a station you would use #3. Set station bits
For that to work SIP needs to be in manual mode. You can switch to manual mode with 1. Change controller values

SlimyFrog123, 2022-06-23:
> OK, I should have pointed you to the GET command reference: https://raw.githubusercontent.com/wiki/Dan-in-CA/SIP/SIP_GET_commands.pdf
>
> For example to show station status you would use # 2. Get [...]

I read the entire GET command reference, and used it whilst making the script.  You can see the script here: https://github.com/SlimyFrog123/SSSC/blob/main/sssc.js

I also know how to use the GET commands, for example in my script, here's the getStatus function:
```
getStatus(stationID, callback = function(data, status) { console.log(data); console.log(status); }) {
  // Get the status of a specific station
  $.get(this.server + encodeURI("sn?sid=" + String(stationID)), function(data, status) {
      callback(data, status);
  });
}
```
https://github.com/SlimyFrog123/SSSC/blob/1a2956605b006f21f495dd1e7060027e518e9112/sssc.js#L96

That function attempts to get the status of a specified station, and it generates and requests the correct link - if I open it in my browser in a new tab, it gives me the value that I am looking for.

Note: the code there is outdated as I have not yet committed the latest version of my program to my GitHub repository, and it is on a different computer.  The only difference between the new code, and the current code in my repository is that I added the pw= parameter in the URL.pa

When I go ahead and open the page, and it requests the page, I see in my console that it could not make the request since, "No Access-Control-Allow-Origin header is present in the requested resource."

Another thing to note is that in your HTTP GET commands reference, it says:

> 2. Get station bits [Keyword /sn]:
> Command: /sn{?sid=x}
> Note: { } means the parameter is optional: if the parameter is not present, the corresponding
> stations name will not be changed. The actual command should not contain the brackets.
> This will return an html webpage that contains a binary value or string that reflects the
> station status.
> Parameters:
> ● sn: returns a list of all stations
> ● sid=x: “x” is the positive station ID (starts from 1).
> Examples:
> http://x.x.x.x/sn (returns the binary status of every station in a string)

Which does not seem to work without the pw={password} parameter, at least not for me.  When I would open the generated link in the browser, it just brought me to the login page, yet when I added the pw={password} parameter, it worked perfectly in the browser, and did not give me the error I was having then, it just gave me the CORS policy error.

Dan-in-CA, 2022-06-23:
As far as the password is concerned you can disable / enable the password under Oprions > Manage Passphrase.
Currently SIP is set to have the passphrase disabled by default.
When the passphrase is disabled you should not need to add it to the GET command.

Can you try adding the following to SIP/webpages.py line 399
web.header(b"Access-Control-Allow-Origin", b"*")
and see if that eliminates the CORS policy error?

Are you running SIP under Python 3?

closed 2024-05-23 by Dan-in-CA