Dan-in-CA/SIP #292: Disable service output to improve SSD lifspan

nmenoni, 2022-08-17:
The `sip.service` output is quite verbose and since it is suppose to run in a RPi where a SSD card is used then we are increasing the chances to kill the SSD card. In order to increase its lifespan I suggest to disable the output adding this option to the **systemd service**:

`StandardOutput=null`

Long term solution is adding a logging mechanism where we can control the logging output severity.

Thanks!

Dan-in-CA, 2022-08-17:
Good suggestion.
Let me do some testing and consider it.
Thanks.

Dan-in-CA, 2022-08-17:
There is a new SIP install script. By default it installs log2ram to reduce SSD writes:
https://github.com/azlux/log2ram

This may be the best solution. What do you think?

nmenoni, 2022-08-18:
That sounds like an interesting approach and can be a good one.
However it has a couple of cons:

1. In case of power off event you will lose everything that wasn't saved yet.
2. You are mounting the entire `/var/log` folder which maybe is not what you want if you have other services running in the same machine and you want to persist all its logs.

Nevertheless, I think that log2ram approach is enough so far.

closed 2022-08-18 by nmenoni