Dan-in-CA/SIP #303: Station Extensions Not Working

Tails86, 2023-05-06:
I updated to 4.1.54 2023-03-25 from a version I think June of last year. The station extensions worked before the update. Then they stopped working after the update.

Dan-in-CA, 2023-05-06:
The latest version of SIP requires Python3.
If you are running an older version of raspberry Pi OS it may not have the RPi.GPIO module for Python 3 installed in which case the stations will not be activated.

Dan

Tails86, 2023-05-06:
I do have Python 3 installed, but it is fairly old now
```
pi@ospi:~ $ python3 --version
Python 3.7.3
```

I did `sudo apt update`, `sudo apt upgrade`, and `sudo apt install python3-rpi.gpio`, but I am still experiencing the same issue with getting zones 9 and 10 to run. Should I try to update my python version?

Dan-in-CA, 2023-05-06:
Python 3.7.3 should be OK.
Are other zones working?
Are you using the Relay board or relay 16 plugin?

Tails86, 2023-05-06:
Zones 1 through 8 work fine. I purchased the zone expander from https://opensprinkler.com/product/zone-expander/ with 16 outputs back in 2017. I'm not using any additional plugin to use this board. I selected that I have "2" station expansions in the Station Handling options, but all I use out of it are zones 9 and 10. Stations 11 through 24 are disabled in the Stations tab.

Dan-in-CA, 2023-05-07:
Testing here seems to be working for zone extensions.
I would suggest going with the db2f59d commit for now.
I will continue to look for a possible bug.

Dan

Tails86, 2023-05-07:
This issue seems to be isolated to the hardware on my end. I took everything apart, and then it started to work fine on my benchtop on master. Something is likely going on with my I2C connection. I swear though that it was pretty consistently not working on master before and consistently working with the previous commit. That's just to say I'm embarrassed 😳 

Thank you for spending the time to look into it! Knowing it worked on your end helped with my diagnosis.

closed 2023-05-07 by Tails86