Dan-in-CA/SIP #309: Wrong log date

ctosae, 2023-09-08:
After the last update the log writes the wrong day

```
{"program": "SP", "station": 2, "duration": "32:33", "start": "23:32:33", "date": "2023-09-07"}
{"program": "SP", "station": 1, "duration": "32:33", "start": "23:00:00", "date": "2023-09-06"}
{"program": "SP", "station": 2, "duration": "32:30", "start": "23:32:30", "date": "2023-09-03"}
{"program": "SP", "station": 1, "duration": "32:30", "start": "23:00:00", "date": "2023-09-03"}
```

Line 1 and 2 are with the new version
Line 3 and 4 are with the old version.

The program start at "23:32:33" (after "23:00:00" +"32:33") so the date is still "2023-09-06" and not "2023-09-07" (when it ends)

Dan-in-CA, 2023-09-08:
Thank you for reporting this.

I will test here and check the code then post here when I find the problem.
Dan

Dan-in-CA, 2023-09-09:
If station 2 starts at "23:32:33" (less than half an hour before midnight) and runs for  32:33" (a little more than half an hour) it would end the next day (after midnight).

The new version of SIP allows for programs to run past midnight.

Dan

ctosae, 2023-09-09:
Also the old program allows to run past midnight (see old log)

The wrong in this version is the (start) time in log.

checking logs it's not clear to understand if it’s the program of the day or of the day before.
Also because the "original" program (station1 + station2) starts at 23.00 of the previous day in log 

I hope you can fix it or I can't understand easly when stations run.

Thanks

Dan-in-CA, 2023-09-09:
OK. 
I see what you are saying. The date field should show when the program started.
I will work on fixing that.

Dan

ctosae, 2023-09-09:
Yes, exactly.

Thank you so much Dan

jonathan-marsh, 2023-09-11:
I noticed this last week and developed a fix, but I haven’t tested it thoroughly yet, or verified that the apparent orphan code I deleted is truly orphaned:

 

It also looks like it affect “node red” programs but I have no idea what that is 😉.  This is not a great fix – the origin of the problem should be that gv.lrun always has a fully accurate time and date stamp for the start time.  But at least it might point to the location of the problem.  

 

-Jonathan

jonathan-marsh, 2023-09-12:
Github stripped the link?  Not so good at git yet.  Here’s a screenshot:

Dan-in-CA, 2023-09-12:
I just pushed an update (version 5.0.18) that I believe fixes 3 time related bugs.

1. the log date bug
2. the program duration bug
3. a bug that would raise an error if a duration for a program was less than 30 seconds. 
I often use 5 or 10 second durations for testing.

Dan

ctosae, 2023-09-12:
> I just pushed an update (version 5.0.18) that I believe fixes 3 time related bugs.

Let me try.... and offer you a pizza and beer in the meantime. Is this still valid? ;-)
https://www.paypal.me/Dan803393

ctosae, 2023-09-13:
Working!

closed 2023-09-15 by ctosae